pibooth/pibooth #182: pibooth 2.0.0: random shutdowns

anxuae, 2020-09-19:
Hi, I was having exactly the same experience as you. I used my PhotoBooth on two weddings so far using Pibooth 2.0.0 and experienced random shutdowns.
Maybe we can share our exact setups (in a structured way) to find similarities.

Raspberry Pi: RPi 4 with 4Gb ram.
Cooling fan on Raspberry Pi: yes
Original RPi USB-C PSU: yes 
Software Version: 2.0.0 (same issues with 1.x.x)
PiBooth LED feature used: No
Pictures additionally stored on USB: Yes
Touchscreen: No
Camera: Canon 2000D
Second camera: No 
Autofocus on camera: Off
Camera mode: Manual
External flash: No
Ferrite bead on camera connection cable: Yes
Printer: DNP DS620
Random crash: approximately after around 150 pictures taken but rather unpredictable.
Crash description: after Liveview (before taking the actual picture) PiBooth quits or goes to the „Uups screen“. 
Does the Pi also crash: No
Solution: restarting the System (power off - power on)

Lets use this structure and please add all things that might be relevant and I will add them in my post as well.

So far, I have added a cable with ferrite bead which seems decrease the crashes.

Greetings from Germany

_Originally posted by @Exitras in https://github.com/pibooth/pibooth/issues/42#issuecomment-695275844_

anxuae, 2020-09-19:
Hello exitras,

I ha e stop using my current pibooth version and I have switched to the current dev version.

Stock with only pibooth, I have tried to reproduce the crash wothout succes.

From now i need to test it with my ds620 to see if it's due to the printing provess. 

So I'll come back on this thread later to inform you if the crash still occur. 

Anxuae informed me that they improved the logs system on the current dev version. 

Maybe I can advise you to test the current dev version in same time of me. 

Papi

_Originally posted by @TheSuperpapi in https://github.com/pibooth/pibooth/issues/42#issuecomment-695302466_

anxuae, 2020-09-19:
Yes maybe but I just tried today,

I'll will add xscreensaver and setup cups later within the day or tomorrow and make new automatic test during 4h00 with rpinting sometime. 

I need to be sure that it'll not crash. 

So, if you find a process to produce the crash, d'o't hesitate. 

On my side to reproduce it on my previous version, I just had to force pibooth to take picture in like- forced mode- via vnc / windows and mouse bot auto click. 

So give me tomorrow and maybe next week to test it a bit more.

_Originally posted by @TheSuperpapi in https://github.com/pibooth/pibooth/issues/42#issuecomment-695321818_

TheSuperpapi, 2020-09-21:
So,

I have good news from now.

Pibooth current dev is running for 2h in photo force mode [bot] and no crash for the moment.

I'll let him run for 4hours more an back here to inform you.

See you,
Papi.

anxuae, 2021-01-31:
Hello @TheSuperpapi , did you have experimented new random shutdown?
Since this issue has not been updated for several months, can we consider that the problem is solved and the issue can be closed?

thanks
anxuae

TheSuperpapi, 2021-01-31:
Hello ans happy New year / Salut a vous et bonne année

Sorry to be off for some weeks now.... Hope i'll spend Time this project again soon...

To answer you, I think it can be close.

Regards and see you soon.

closed 2021-04-23 by anxuae