pibooth/pibooth #183: Fresh install with raspberry pi OS and pibooth 2.0, not working out of the box

DJ-Dingo, 2020-09-21:
I downloadet the new Raspberry Pi OS, update & upgrade, and installed Pibooth 2.0
When i wanna go to pibooth --config it says error. (See log)
So i used - "sudo python3 -m pip install -U pygame --user" to get it working. But i can't use pibooth --config from command promt, i have to do it manually, sudo nano ~/.config/pibooth/pibooth.cfg or from desktop - When i use pibooth --config it is open fine but is not reacting, and making all kind of wierd letters in the file when you press the keyboard. I normally use nano and choose that as the default, but it sometimes shows up with vim?

On my other SD card (Old Raspbian) it automatic pops up with "Mousepad" for edit, when you use pibooth --config, i like that, how do i turn that on to do that?.

Here is the install info, it says something with Pillow
-------------------------------------------------------------------
sudo pip3 install pibooth
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pibooth
  Downloading https://files.pythonhosted.org/packages/0f/35/eb070ca2959e384e0ebd593cb5c524b1866192722119234d9f95cd08b1c4/pibooth-2.0.0-py2.py3-none-any.whl (715kB)
    100% |████████████████████████████████| 716kB 264kB/s 
Collecting pluggy>=0.13.1 (from pibooth)
  Downloading https://files.pythonhosted.org/packages/a0/28/85c7aa31b80d150b772fbe4a229487bc6644da9ccb7e427dd8cc60cb8a62/pluggy-0.13.1-py2.py3-none-any.whl
Requirement already satisfied: psutil>=5.5.1 in /usr/lib/python3/dist-packages (from pibooth) (5.5.1)
Requirement already satisfied: Pygame>=1.9.4 in /usr/lib/python3/dist-packages (from pibooth) (1.9.4.post1)
Requirement already satisfied: picamera>=1.13; platform_machine >= "armv0l" and platform_machine <= "armv9l" in /usr/lib/python3/dist-packages (from pibooth) (1.13)
Collecting pygame-menu>=3.0.1 (from pibooth)
  Downloading https://files.pythonhosted.org/packages/46/e1/0e2a5725ac18f939d254e603afed6c4fb02cffd8b51b06bd378728a7e2db/pygame_menu-3.1.4-py2.py3-none-any.whl (952kB)
    100% |████████████████████████████████| 962kB 216kB/s 
Requirement already satisfied: gpiozero>=1.5.1 in /usr/lib/python3/dist-packages (from pibooth) (1.5.1)
Collecting Pillow>=6.1.0 (from pibooth)
  Downloading https://www.piwheels.org/simple/pillow/Pillow-7.2.0-cp37-cp37m-linux_armv7l.whl (1.3MB)
    100% |████████████████████████████████| 1.3MB 150kB/s 
Collecting importlib-metadata>=0.12; python_version < "3.8" (from pluggy>=0.13.1->pibooth)
  Downloading https://files.pythonhosted.org/packages/8e/58/cdea07eb51fc2b906db0968a94700866fc46249bdc75cac23f9d13168929/importlib_metadata-1.7.0-py2.py3-none-any.whl
Collecting pyperclip (from pygame-menu>=3.0.1->pibooth)
  Downloading https://www.piwheels.org/simple/pyperclip/pyperclip-1.8.0-py3-none-any.whl
Collecting zipp>=0.5 (from importlib-metadata>=0.12; python_version < "3.8"->pluggy>=0.13.1->pibooth)
  Downloading https://files.pythonhosted.org/packages/b2/34/bfcb43cc0ba81f527bc4f40ef41ba2ff4080e047acb0586b56b3d017ace4/zipp-3.1.0-py3-none-any.whl
pygame-menu 3.1.4 has requirement pygame>=1.9.6, but you'll have pygame 1.9.4.post1 which is incompatible.
Installing collected packages: zipp, importlib-metadata, pluggy, pyperclip, pygame-menu, Pillow, pibooth
  Found existing installation: Pillow 5.4.1
    Not uninstalling pillow at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'Pillow'. No files were found to uninstall.
Successfully installed Pillow-7.2.0 importlib-metadata-1.7.0 pibooth-2.0.0 pluggy-0.13.1 pygame-menu-3.1.4 pyperclip-1.8.0 zipp-3.1.0

Here the pibooth --config info, before i manually updated pygame 1.9.4 with "sudo python3 -m pip install -U pygame --user" to ver 1.9.6
--------------------------------------------------------------------------------------------------
pibooth --config
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 6, in <module>
    from pibooth.booth import main
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 27, in <module>
    from pibooth.config import PiConfigParser, PiConfigMenu
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/__init__.py", line 4, in <module>
    from pibooth.config.menu import PiConfigMenu
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/menu.py", line 7, in <module>
    import pygame_menu as pgm
  File "/usr/local/lib/python3.7/dist-packages/pygame_menu/__init__.py", line 66, in <module>
    import pygame_menu.themes
  File "/usr/local/lib/python3.7/dist-packages/pygame_menu/themes.py", line 404, in <module>
    THEME_DEFAULT = Theme()
  File "/usr/local/lib/python3.7/dist-packages/pygame_menu/themes.py", line 202, in __init__
    bool, pygame.vernum.major == 2)  # type: bool
AttributeError: 'tuple' object has no attribute 'major'

anxuae, 2020-09-23:
Hello @DJ-Dingo ,

thanks for report, I need to release a new version to fix it.
In fact, pibooth is compatible with `pygame == 1.9.6`.

Note: when you install a python library, it is easier to run:

```
pip3 install -U pygame
```

instead of 

```
python3 -m pip install -U pygame
```

Note2: is you use the `--user` option, please don't use `sudo`:

```
'sudo' is use to install programs as super-user, that means for all users which can be login in to your raspberry-pi. Using the option `--user` means install the program in the home directory of the user who install it. If you combine the both, you will create files in your home folder with super-user rights, thus will have rights access issue each time you launch your program.
```

anxuae, 2020-09-23:
Version 2.0.1 released! It should be fixed now.
Please reopen if not.

closed 2020-09-23 by anxuae