pibooth/pibooth #184: Wrong image(s) used when processing images of Sony (A5100)

Eddy-1980, 2020-09-24:
When using Pibooth in combination with Sony A5100 the wrong image has been used during processing.

1) Click on button to take 1 image
2) liveview of Sony A5100 works
3) Photo has been taken
4) Processing
5) Photocard with wrong image. Pibooth uses the photo taken one or two sessions before

I will upload video.

anxuae, 2020-09-24:
Hello @Eddy-1980,
Can you start pibooth in a shell with the `-v` option ?

Then paste the content of the shell when the `oops` happen.

Thanks

botyas, 2020-09-24:
Good night to me with a canon 450D the same thing happens to me this is what the console shows:

[ DEBUG   ] pibooth           : Activate state 'processing'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ INFO    ] pibooth           : +< Saving raw captures
[ DEBUG   ] pibooth           : +> took 0.000 seconds
[ ERROR   ] pibooth           : [-1] Unspecified error
Traceback (most recent call last):
  File "/home/pi/pibooth/pibooth/booth.py", line 362, in main_loop
    self._machine.process(events)
  File "/home/pi/pibooth/pibooth/states.py", line 50, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/pibooth/pibooth/plugins/picture_plugin.py", line 100, in state_processing_do
    captures = app.camera.get_captures()
  File "/home/pi/pibooth/pibooth/camera/base.py", line 92, in get_captures
    images.append(self._post_process_capture(data))
  File "/home/pi/pibooth/pibooth/camera/gphoto.py", line 124, in _post_process_capture
    camera_file = self._cam.file_get(gp_path.folder, gp_path.name, gp.GP_FILE_TYPE_NORMAL)
gphoto2.GPhoto2Error: [-1] Unspecified error
[ ERROR   ] pibooth           : system='Linux', node='photobooth', release='5.4.51-v7+', version='#1333 SMP Mon Aug 10 16:45:19 BST 2020', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

anxuae, 2020-09-26:
Hi @botyas 

Thanks for report, can you attach the log file `/tmp/pibooth.log` ?

Can you check that you have not configured your camera with JPEG+RAW capture format?

botyas, 2020-09-27:
good morning here I leave the log file.
1- try 4 shots with error
2- test 1 take well

Rapsberry Pi 3 B+
Canon 450D - no card
Printer Canon Shelpy CP1200 - not configured
Led Strib RGB

Thanks for your help and your work

[pibooth.log](https://github.com/pibooth/pibooth/files/5287913/pibooth.log)

anxuae, 2020-09-27:
Hello @botyas ,
Can you post the log generated by the command:

```
pibooth-diag
```

Don't see the reason why, the image download from the DSLR seems blocked.

botyas, 2020-09-27:
[diagnostic.log](https://github.com/pibooth/pibooth/files/5289106/diagnostic.log)

werdeil, 2020-09-28:
From what I see the issue come from the fact that you don't have a SD card, the camera can save one photo on its internal memory but not 4 of them, hence the issue on the download.

@botyas can you try with a sd card in the camera?

@Eddy-1980 do you use an SD card on your setup?

Eddy-1980, 2020-09-28:
@werdeil yes I'm using a SD card into my Sony A5100 camera.

anxuae, 2020-09-28:
Bad news for Sony camera: https://github.com/gphoto/gphoto2/issues/117#issuecomment-364077785

mattdsteele, 2021-03-27:
As info, I'm experiencing the same issue. I can capture a single photo successfully, but on the 4-photo flow, after taking the 4th photo it fails and goes to the failsafe state.

* Raspberry Pi 4
* Sony 6400
* SD card installed
* gphoto2-2.5.27
* libgphoto2-2.5.22
Logs:

```
[ DEBUG   ] pibooth           : Activate state 'capture'
[ INFO    ] pibooth           : +< Take a capture
[ DEBUG   ] pibooth           : +> took 2.499 seconds
[ DEBUG   ] pibooth           : Activate state 'processing'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ INFO    ] pibooth           : +< Saving raw captures
[ DEBUG   ] pibooth           : +> took 0.000 seconds
[ ERROR   ] pibooth           : [-1] Unspecified error
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 50, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/picture_plugin.py", line 100, in state_processing_do
    captures = app.camera.get_captures()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/base.py", line 92, in get_captures
    images.append(self._post_process_capture(data))
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 131, in _post_process_capture
    camera_file = self._cam.file_get(gp_path.folder, gp_path.name, gp.GP_FILE_TYPE_NORMAL)
gphoto2.GPhoto2Error: [-1] Unspecified error
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1403 SMP Mon Feb 22 11:33:35 GMT 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : Activate state 'wait'

```

anxuae, 2021-05-11:
Hi @mattdsteele ,

Can you attache the file generated with the command following command?

```
pibooth-diag
```

(your DSLR shall be ON and connected to the raspberry pi)

anxuae, 2021-05-11:
I'm still trying to find an alternative, but I need to test on a Sony to see what are the alternatives (if any)

mattdsteele, 2021-05-14:
@anxuae I've attached my diagnostics log file: 
[diagnostic.log](https://github.com/pibooth/pibooth/files/6480602/diagnostic.log)

The console output a large number of warnings and errors when running `pibooth-diag`, primarily these:

```
Gphoto2: b'ptp_usb_event [usb.c:588]': b'Reading PTP event failed: Timeout reading from or writing to the port (-10)'
Gphoto2: b'ptp_usb_event [usb.c:588]': b'Reading PTP event failed: Timeout reading from or writing to the port (-10)'
Gphoto2: b'camera_sony_capture [library.c:4808]': b'no object found during event polling. perhaps no focus...'
Gphoto2: b'gp_camera_capture [gphoto2-camera.c:1340]': b"'camera->functions->capture (camera, type, path, context)' failed: -1"
```

anxuae, 2021-05-14:
Hi @mattdsteele,

Before restarting `pibooth`, can you run the following command in a shell:

```
gphoto2 --set-config capturetarget="card+sdram"
```

And tell me if you can now take at least one capture?

anxuae, 2021-05-14:
Also I see that your DSLR is setup with "RAW+JPEG (X.Fine)": two formats is not supported by `pibooth`, please choose only between Fine or Extra Fine.

anxuae, 2021-05-14:
@Eddy-1980 cat you post the file generated by the `pibooth-diag` command (with your Sony A5100 connected)?
I'm trying to find likeness between different Sony models.

Thanks

mattdsteele, 2021-05-14:
Thanks @anxuae. I'm not sure what's changed between March and now, so I appreciate your help in debugging!

> Before restarting pibooth, can you run the following command in a shell:

I wasn't sure the specific ordering you're looking for, so I tried the following:

1. Launch pibooth
2. Execute `gphoto2 --set-config capturetarget="card+sdram"` in a separate terminal

At this point it fails with the following error:

```
pi@raspberrypi:~ $ gphoto2 --set-config capturetarget="card+sdram"
                                                                               
*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
```

I then tried to run it without pibooth open:

```
pi@raspberrypi:~ $ gphoto2 --set-config capturetarget="card+sdram"
                                                                               
*** Error ***              
capturetarget not found in configuration tree.
*** Error (-1: 'Unspecified error') ***    
```

Here's the logs with the `--debug` flag active: [set-config-debug.log](https://github.com/pibooth/pibooth/files/6481139/set-config-debug.log)

> Also I see that your DSLR is setup with "RAW+JPEG (X.Fine)": two formats is not supported by pibooth, please choose only between Fine or Extra Fine.

Nice catch - unfortunately I've switched the capture to "JPEG Only (Fine)" and am getting the same result.

anxuae, 2021-05-15:
Ok, let it do also manually: can you switch from 'sdram' to 'card+sdram' and tell me if it's working better?

mattdsteele, 2021-05-16:
Running this returned a similar error:

```
pi@raspberrypi:~ $ gphoto2 --set-config capturetarget="sdram"
                                                                               
*** Error ***              
capturetarget not found in configuration tree.
*** Error (-1: 'Unspecified error') ***       
```

I've attached the logs; they appear to be pretty similar to `card+sdram`. 
[set-config-debug-sdram.log](https://github.com/pibooth/pibooth/files/6488702/set-config-debug-sdram.log)

(As an aside, I've been successful using a standard Logitech webcam with this pibooth instance, which is my backup plan if we can't get the DSLR working)

Eddy-1980, 2021-05-16:
At this moment I'm not able to make any photo with pibooth and Sony A5100. When I try to start a session pibooth show me "OOPS something went wrong". So trying to figure out which settings I have changed.

anxuae, 2021-05-16:
Hi @Eddy-1980 and @mattdsteele 

Can you try to use pibooth with the development version? Here is the commands to execute in a shell:

```
$> git clone https://github.com/pibooth/pibooth.git
$> cd pibooth
$> git checkout camera
$> git pull
$> pip install -e .[dslr,printer]
$> pibooth
```

(those commands will install the unnstall the `pibooth` official version)

Eddy-1980, 2021-05-16:
@anxuae 

This is what I have done:
RPi4 connected with Sony A5100
used development version

(1)
- Start pibooth
- Start photo session with 1 photo
- Processing / Result 1 photo / Session successfull
- Start 2nd session with 1 photo
- Processing / Result 1 photo, this photo is turned 90degrees,  photo of the 1st session has been used for processing
- Start 3rd session with 1 photo
- Processing / Result 1 photo,  photo of the 2nd session has been used for processing
- etc
- after a couple of session I get the messages "Oops something went wrong"

(2)
- Start pibooth
- Start photo session with 4 photo
- 1 photo is taken, thereafter I get the message:  "Oops something went wrong"
[pibooth-1photo-developmentversion.log](https://github.com/pibooth/pibooth/files/6489751/pibooth-1photo-developmentversion.log)
[pibooth-4photos-developmentversion.log](https://github.com/pibooth/pibooth/files/6489752/pibooth-4photos-developmentversion.log)

mattdsteele, 2021-05-16:
I'm seeing similar things using the development version.

Run 1: [a6400-run-1.log](https://github.com/pibooth/pibooth/files/6489950/a6400-run-1.log)

* Start pibooth
* Start first session with 1 photo
* Shutter goes off twice
* Processing / Session successful
* Start second session with 1 photo
* Shutter goes off once
* Processing / Result shows photo taken during first session
* Start third session with 1 photo
* Shutter goes off twice
* Processing / Result shows photo taken during second session
* etc. I wasn't able to trigger the error page, and quit after about 10 sessions.

Run 2: [a6400-run-2.log](https://github.com/pibooth/pibooth/files/6489951/a6400-run-2.log)

* Start pibooth
* Start run with 4 photos
* 4 photos are taken (each time the shutter goes off twice), then "Oops something went wrong" page.

In addition, here's the output from `pibooth-diag` (which also triggered two shutters when capturing): [diagnostic.log](https://github.com/pibooth/pibooth/files/6489956/diagnostic.log)

mattdsteele, 2021-05-16:
I found my shooting mode was set to "Continuous - LO", which I think was responsible for the multiple shutter triggers. After switching to "Single" shooting mode, here's a few more results:

Run 3: [a6400-run-3.log](https://github.com/pibooth/pibooth/files/6489984/a6400-run-3.log)

* Start pibooth
* Successful 1 photo session (shutter goes off once, displays photo from current session)
* Successful 1 photo session (shutter goes off once, displays photo from current session)
* Successful 1 photo session (shutter goes off once, displays photo from current session)
* Successful 1 photo session (shutter goes off once, displays photo from current session)
* Unsuccessful session (Session counts down to "Smile", then does not actually take photo, and about 30 seconds later displays "Something went wrong" page.

Run 4: [a6400-run-4.log](https://github.com/pibooth/pibooth/files/6489991/a6400-run-4.log)

* Start pibooth
* Start session with 4 photos
4 photos are taken (each time the shutter goes off once), then "Oops something went wrong" page.

Let me know if there are any other tests you'd like me to run, thanks.

anxuae, 2021-05-17:
Hi @mattdsteele,

Thanks a lot for your valued debuting information. I have implemented a fix for Sony DSLR which consists in getting captures just after each shot instead of waiting for the `processing` state.

Can you try it?
```bash
$> git pull
$> git checkout sony
```

Reset the configuration, you should see a new parameter `download_after_capture` in `[CAMERA]` section.
Set it to `True` then start pibooth:

```bash
$> pibooth
```

Eddy-1980, 2021-05-17:
Hi @anxuae 

This is what I get when I start a session with 4 photos.
[pibooth-4photos-developmentversion.log](https://github.com/pibooth/pibooth/files/6496818/pibooth-4photos-developmentversion.log)

mattdsteele, 2021-05-18:
I think we're making progress! After checking out the `sony` branch and re-installing, I'm able to successfully run a 4-capture session 😄 

I am still seeing I/O timeout errors after running a few pibooth sessions; these manifest as a 30 second delay on the "Smile" screen, followed by the "something went wrong" error screen.

As an example, here's the logs from a run where I successfully did a 4-capture session, followed by an unsuccessful 1-capture session: 
[sony-branch-session-1.log](https://github.com/pibooth/pibooth/files/6497910/sony-branch-session-1.log)

Eddy-1980, 2021-05-18:
Hi @anxuae @mattdsteele 

I'm using the same development version as matt. But I can not make 4 photos. 
After the first has been taken the system crashed.
I'm using the Sony A5100

anxuae, 2021-05-18:
@mattdsteele 

I pushed a dirty hack which try a second time to get the capture if the first time failed. However it's seems we have reached the limit of the camera. Seen those lines in yours logs, let me thinking that the PTP link with the camera is not really strong;

```bash
'libusb_clear_halt(port->pl->dh, internal_ep)' failed: Other error (-99)
'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Input/Output error (-1)
PTP_OC 0x1008 receiving data failed: PTP General Error (0x2002)
'ptp_getobjectinfo (params, newobject, &oi)' failed: PTP I/O Error (0x02ff)
```

@Eddy-1980 

Did you checkout the `sony` branch? It is strange because `pibooth` crashes at the beginning of the second preview with a `PTP Access Denied (0x200f)` error: it seems that pibooth has not anymore access to the camera: seems to be a crash of the camera software...
Does it be systematic?

mattdsteele, 2021-05-19:
@anxuae Pulling the most recent changes, I'm seeing relatively similar results. I don't see anything in the logs indicating a second capture attempt.

The error does appear different (`PTP General Error`).

<details><summary>Recent logs (click to expand)</summary>

```
[ DEBUG   ] pibooth           : Activate state 'capture'
[ INFO    ] pibooth           : +< Take a capture
[ DEBUG   ] pibooth.gphoto2   : |  camera_sony_capture [library.c:4215]: no object found during event polling. try the 0xffffc001 object id
[ DEBUG   ] pibooth.gphoto2   : |  gp_libusb1_read [libusb1.c:610]: 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Input/Output error (-1)
[ DEBUG   ] pibooth.gphoto2   : |  gp_port_read [gphoto2-port.c:441]: Reading 1024 = 0x400 bytes from port failed: Error reading from the port (-34)
[ DEBUG   ] pibooth.gphoto2   : |  gp_libusb1_clear_halt_lib [libusb1.c:582]: 'libusb_clear_halt(port->pl->dh, internal_ep)' failed: Other error (-99)
[ DEBUG   ] pibooth.gphoto2   : |  gp_libusb1_read [libusb1.c:610]: 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Input/Output error (-1)
[ DEBUG   ] pibooth.gphoto2   : |  gp_port_read [gphoto2-port.c:441]: Reading 1024 = 0x400 bytes from port failed: Error reading from the port (-34)
[ DEBUG   ] pibooth.gphoto2   : |  gp_libusb1_clear_halt_lib [libusb1.c:582]: 'libusb_clear_halt(port->pl->dh, internal_ep)' failed: Other error (-99)
[ DEBUG   ] pibooth.gphoto2   : |  ptp_usb_getdata [usb.c:479]: PTP_OC 0x1008 receiving data failed: PTP General Error (0x2002)
[ DEBUG   ] pibooth.gphoto2   : |  camera_sony_capture [library.c:4219]: 'ptp_getobjectinfo (params, newobject, &oi)' failed: PTP I/O Error (0x02ff)
[ DEBUG   ] pibooth.gphoto2   : |  gp_camera_capture [gphoto2-camera.c:1340]: 'camera->functions->capture (camera, type, path, context)' failed: -7
[ WARNING ] pibooth           : |  Gphoto2 fails to capture, trying again...
[ DEBUG   ] pibooth.gphoto2   : |  gp_libusb1_write [libusb1.c:596]: 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Input/Output error (-1)
[ DEBUG   ] pibooth.gphoto2   : |  gp_port_write [gphoto2-port.c:406]: Writing 12 = 0xc bytes to port failed: Error writing to the port (-35)
[ DEBUG   ] pibooth.gphoto2   : |  ptp_usb_sendreq [usb.c:109]: PTP_OC 0x9209 sending req failed: Error writing to the port (-35)
[ DEBUG   ] pibooth.gphoto2   : |  gp_libusb1_clear_halt_lib [libusb1.c:582]: 'libusb_clear_halt(port->pl->dh, internal_ep)' failed: Other error (-99)
[ DEBUG   ] pibooth.gphoto2   : |  gp_libusb1_write [libusb1.c:596]: 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Input/Output error (-1)
[ DEBUG   ] pibooth.gphoto2   : |  gp_port_write [gphoto2-port.c:406]: Writing 12 = 0xc bytes to port failed: Error writing to the port (-35)
[ DEBUG   ] pibooth.gphoto2   : |  ptp_usb_sendreq [usb.c:109]: PTP_OC 0x9209 sending req failed: Error writing to the port (-35)
[ DEBUG   ] pibooth.gphoto2   : |  camera_wait_for_event [library.c:5860]: 'ptp_sony_getalldevicepropdesc (params)' failed: PTP General Error (0x2002)
[ DEBUG   ] pibooth.gphoto2   : |  gp_camera_wait_for_event [gphoto2-camera.c:1451]: 'camera->functions->wait_for_event ( camera, timeout, eventtype, eventdata, context)' failed: -1
[ DEBUG   ] pibooth           : +> took 35.932 seconds
[ ERROR   ] pibooth           : [-1] Unspecified error
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/home/pi/code/pibooth/pibooth/states.py", line 50, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/code/pibooth/pibooth/plugins/camera_plugin.py", line 102, in state_capture_do
    app.camera.capture(effect)
  File "/home/pi/code/pibooth/pibooth/camera/gphoto.py", line 324, in capture
    ev_type, ev_data = self._cam.wait_for_event(1000)
[... cut ...]
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1414 SMP Fri Apr 30 13:20:47 BST 2021', machine='armv7l', processor=''
```
</details>

I think you're correct that the issue may be with the connection, though. Switching to a different USB cable (and connecting directly to the Pi instead of through a USB hub), I'm able to get slightly more stability, but even then, after about 10 sessions, it eventually times out while trying to take a capture.

At this point I'm ready to declare victory, and maybe focus on investigating the issue with the gphoto2 team. Thanks again for all your help trying to debug this.

anxuae, 2021-05-19:
Thanks for your feedback. Wahoo, the capture global time took 35s, quite long... The retry has been done, see the line:
```
[ WARNING ] pibooth           : |  Gphoto2 fails to capture, trying again...
```
However it has also failed. Yes is is probably an issue of the camera that gphoto2 does not handles. A last try could be to reset the connection with the camera after each capture, not really conventional be we can try.

mattdsteele, 2021-05-19:
I appreciate the effort @anxuae but I think rather than encode further workarounds for a flaky device, I'll just find a friend with a Canon or Nikon DSLR and borrow it for my use case 😅

mattdsteele, 2021-05-20:
A quick update: on a lark, I tried installing pibooth on another machine (Mac OS X 10.15.7), and it's not completely perfect, but it does seem much more stable. I've only seen a timeout once (out of about 100 captures), and the retry logic was successful.

adimajo, 2023-08-17:
Hi,
Seems like a very similar issue so since the issue is still open, I figured I could add on here; feel free to tell me to open another issue.

Sony alpha 7R2 here.

pibooth boots, takes 1 picture (whether in 1- or 4-photo(s) mode, what I mean by that is that if I hit the 1 picture button, then retry, it will crash as well). Logs:

```
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video0'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src1 reported: Cannot identify device '/dev/video1'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src2 reported: Cannot identify device '/dev/video2'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video2): can't open camera by index
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ ERROR   ] pibooth           : Unsupported option settings/capturetarget=Memory card ([-2] Bad parameters), configure your DSLR manually
[ WARNING ] pibooth           : No printer configured in CUPS (see http://localhost:631)
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ ERROR   ] pibooth           : [-7] I/O problem
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.21-v7+', version='#1642 SMP Mon Apr  3 17:20:52 BST 2023', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
```

The trick with gphoto2 didn't work as well:

```
gphoto2 --set-config capturetarget="sdram"
                                                                               
*** Error ***              
capturetarget not found in configuration tree.
*** Error (-1: 'Unspecified error') ***       
```

Also, I cannot reconnect the camera in "PC mode" after a crash, only "USB mode" and I have to reboot the Pi.

Output of `pibooth-diag`:

```

```

bryanvf, 2023-11-13:
I also have the same problem.
Is there any update on this topic?