pibooth/pibooth #185: Can i Install plugins, that don´t effect the running of PiBooth

DJ-Dingo, 2020-09-28:
I have made a plugin where an LCD screen showing "photos taken - time/date". and It is all working fine.
But i found out that if the external power supply to the LCD goes out, or a wire jumps off the LCD, then PiBooth shuts down.

Would it be possible to install some of your plugins in a way that do not effect or shut down pibooth, when these "hardware" errors occur. ? It is just a screen that´s not important to run pibooth, but just a nice extra future to have in the front.

werdeil, 2020-09-28:
It is a choice to make pibooth crash if the plugin crashes, however you can avoid your issue by handling the exceptions in the plugin.

Do you have a repository for this plugin? I could take an look and try to help.

DJ-Dingo, 2020-09-28:
Thanks @werdeil 

Yes, i just uploadet it. This is my first time using Github like this, so not shure if i done it right. :-) 
[https://github.com/DJ-Dingo/Pibooth-LCD-with-I2c](url)

werdeil, 2020-09-28:
Perfect! I'll have a look on giving you a possible fix, can you tell me what kind of error is raised when the display is not available?

Can you run the pibooth in debug mode, cut the power of the display and show me the traceback?

DJ-Dingo, 2020-09-28:
shure. Whats the command for debug  --D or ?

werdeil, 2020-09-28:
It is an option in the config, you can access it directly in via the menu in pibooth or the `pibooth --config` command

DJ-Dingo, 2020-09-28:
Ohh yes.. (Just forgot to copy my photos first, so it deleted them) :-(

I have 2 LCD on at the moment, It's the one with I2c that i am power of "lcd_i2c_plugin-1.0.1"

pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: piboothLCD-1.0.1, extra-lights-1.0.0, lcd_i2c_plugin-1.0.1, sound-effects-0.0.3
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 361, in main_loop
    self._machine.process(events)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 51, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/lcd_i2c_plugin.py", line 52, in state_wait_do
    app.lcd.write_string('%s' % time.strftime('%d/%m - %H:%M:%S'))
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 283, in write_string
    self.write(char)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 410, in write
    self.cursor_pos = newpos
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 158, in _set_cursor_pos
    self.command(c.LCD_SETDDRAMADDR | row_offsets[value[0]] + value[1])
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 381, in command
    self._send_instruction(value)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/i2c.py", line 241, in _send_instruction
    (value & 0xF0)) | self._backlight)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 439, in main
    app.main_loop()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 367, in main_loop
    self._pm.hook.pibooth_cleanup(app=self)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/piboothLCD.py", line 111, in pibooth_cleanup
    app.lcd.clear()
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 311, in clear
    self.command(c.LCD_CLEARDISPLAY)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 381, in command
    self._send_instruction(value)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/i2c.py", line 241, in _send_instruction
    (value & 0xF0)) | self._backlight)
OSError: [Errno 121] Remote I/O error

DJ-Dingo, 2020-09-28:
Should i disconnect the other LCD, so it dont give info about that one. Just to make it more simpel for you?

DJ-Dingo, 2020-09-28:
Here the info without the other LCD ( have a LCD without the I2c, for testing that too )

pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: extra-lights-1.0.0, sound-effects-0.0.3, lcd_i2c_plugin-1.0.1
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 361, in main_loop
    self._machine.process(events)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 51, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/lcd_i2c_plugin.py", line 51, in state_wait_do
    app.lcd.cursor_pos = (1, 0)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 158, in _set_cursor_pos
    self.command(c.LCD_SETDDRAMADDR | row_offsets[value[0]] + value[1])
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 381, in command
    self._send_instruction(value)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/i2c.py", line 241, in _send_instruction
    (value & 0xF0)) | self._backlight)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 439, in main
    app.main_loop()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 367, in main_loop
    self._pm.hook.pibooth_cleanup(app=self)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/lcd_i2c_plugin.py", line 106, in pibooth_cleanup
    app.lcd.close(clear=True)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 142, in close
    self.clear()
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 311, in clear
    self.command(c.LCD_CLEARDISPLAY)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/lcd.py", line 381, in command
    self._send_instruction(value)
  File "/usr/local/lib/python3.7/dist-packages/RPLCD/i2c.py", line 241, in _send_instruction
    (value & 0xF0)) | self._backlight)
OSError: [Errno 121] Remote I/O error

werdeil, 2020-09-28:
No need, have look at your plugin GitHub, I just pushed a PR that you can try

And sorry I forgot to warn you about the deletion of pictures in debug mode...

DJ-Dingo, 2020-09-28:
Thx alot. That seems to work. :-) 
The sec.in the clock is jumping(stopping) a little more doing the photo take, but i can live with that. :-)

This is what happens when i pull the power and put it back, while pibooth is still running.
When i put back power i get a blank screen, with only the numbers of photo taken, but not the text (normally i could just take a photo or go in the config and out and the text came back, if there was a mistake on the screeen). 
Now i only see every 10 sec as 1, 2. 3 esc, and minutes as 1,2,3,4,5 etc. Can we have it rewrite the text and clock to the screen, if power is of, and on.

werdeil, 2020-09-28:
It seems that the `write_string` command only refresh the characters that change and don't write everything (as I would have expected.

I've re added a `clear()` each time we refresh the photo count, it should behave as before now (refresh when going in the config or take a photo).

For the clock jumping more I don't see why it changed but I'll have a look.

DJ-Dingo, 2020-09-28:
Ok. it's just sec. stopping between each sequence, taken a photo. not that important.

Ok i give it a try :-)

Another thing, Do I just need to press "Merge Pull Request" to make your changes to my github. Still new to this.

werdeil, 2020-09-28:
Yes if you agree with my PR you can press "Merge Pull Request"

DJ-Dingo, 2020-09-28:
It is working perfect now :-) Thanks @werdeil 

So to make this a generic LCD_I2c_plugin, the next thing should be a setup.py  ?
And the possibility in pibooth --config to change "Text", "Port Expanders", and port "Address"

( And maybe in the future to have the possibilities to use app.count.printed, app.count.forgotten, app.count.remaining_duplicates as a choice instead of etc. date/time in a 16x2 LCD) I am still waiting for my 16x4 display that can show 4 lines, but it seems to be month before it get's here.And i don't use Print, so not that important to me.)

Text chance in config "Today Photos" (With on my display it is max 12 (16 " -4" for the counter and space), i don't have a 20x2 LCD yet, so can't test that. (could offcouse use rolling text for more text)
 - app.lcd.write_string('Today Photos %s' % app.count.taken)

Port Expander and Address
 - app.lcd = CharLCD('PCF8574', 0x3F)

--Config
[LCD_WITH_I2C]

# Text before "number of photos taken" Max 12 with a 16x2 LCD
Text = "Photos Taken"

# Supported port expanders are the (PCF8574 -Default), the MCP23008 and the MCP23017.
Port Expander = PCF8574

# I2c port address (Default 0x27)
Address = 0x3F

werdeil, 2020-09-28:
I've open a discussion directly on the plugin repo, I think we can close this one on pibooth.

DJ-Dingo, 2020-09-28:
Yes, lets do that :-)

closed 2020-09-28 by DJ-Dingo