pibooth/pibooth #186: Slow problem with a webcam

babou7635, 2020-10-06:
Hello,
I come back to you for a new question. When I run Pibooth with a webcam the preview is crazy (3 seconds between the real image and the preview). And the countdown is no longer displayed with numbers but with transparent rectangles.
While when I use the Camera of the raspberry (V2 black) there was no problem.

**Environment**
 - OS Raspbian
 - Camera Webcam papalook PA552
 - Pibooth version V2.0.0

anxuae, 2020-10-06:
Hi @babou7635 

Thanks for reporting. It is probably the same issue than #91. 
Your camera seems to have a very high resolution, and processing each frame takes time.

Code optimization may be done in the following method:

https://github.com/pibooth/pibooth/blob/2e538441ae4c129c44b4b17c9d7337700e39181e/pibooth/camera/opencv.py#L77-L102

But I have no such camera to do this work

werdeil, 2020-10-14:
Hi @babou7635,

About the transparent rectangle: can you tell us which version of Pillow you use (just write `pip show pillow` on a shell)

If it is version 7.2.0, try to downgrade to 7.1.2 (`pip install pillow=7.1.2`) and tell us if it solves at least the transparent rectangles.

About the slowness I'm trying to reproduce it, I had the same issue but I can't reproduce now, I need to understand why.

werdeil, 2020-10-16:
hi @babou7635, I've reproduce the slowness problem with my Logitech C920 and a Raspberry Pi 3B, and I pushed a PR (#193) that could a workaround on this.

mmk8283, 2020-10-25:
I am also having issues with the white transparent rectangles appearing instead of countdown numbers. This happens when using a webcam, if I use a pi camera the numbers appear correctly. I tried different versions of Pillow with no change. Any help on this issue would be appreciated 

Thank you!

anxuae, 2020-11-14:
Hi everybody,

The PR #193 is now merged, @babou7635 you can try if the preview is improved by installing a developing version of pibooth.

anxuae, 2021-01-31:
Since no news about this issue, I consider the fix PR #193 is solving it.
Please feel free to re-open this issue if it is not the case.

closed 2021-01-31 by anxuae