pibooth/pibooth #187: Mostrar vista previa antes de la captura

AcidCool83, 2020-10-07:
Good,

I have managed to configure the application to my liking but I am unable to make the preview image of the camera appear before taking the capture. I am using the PiCamara. Please, could you help me to see the image in the moment before taking the photo?

Thank you very much and greetings

anxuae, 2020-10-11:
Hello @AcidCool83 

Can you add generated log file and your config file?

anxuae, 2020-10-12:
Which king of screen are you using?

AcidCool83, 2020-10-13:
Hi @anxuae 

The screen is 19 inch.

This is the log text:
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: sound-effects-0.0.3, extra-lights-1.0.0
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : BUTTONDOWN: generate CAPTURE button event
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : +< Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : +> took 1.000 seconds
[ DEBUG   ] pibooth           : BUTTONDOWN: generate CAPTURE button event
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : +< Show picture choice (4 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[... 246 lines cut ...]
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to draw texts
[ DEBUG   ] pibooth           : |  +> took 3.200 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.105 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to outline boundary borders
[ DEBUG   ] pibooth           : |  +> took 0.002 seconds
[ DEBUG   ] pibooth           : |  +> took 0.129 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to outline boundary borders
[ DEBUG   ] pibooth           : |  +> took 0.003 seconds
[ DEBUG   ] pibooth           : |  +> took 4.200 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.304 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to outline boundary borders
[ DEBUG   ] pibooth           : |  +> took 0.002 seconds
[ DEBUG   ] pibooth           : Add to buffer the image '1911138064'

And I attach my configuration file
[pibooth.txt](https://github.com/pibooth/pibooth/files/5372667/pibooth.txt)

Thanks

AcidCool83, 2020-10-15:
@anxuae please help me

anxuae, 2020-10-15:
Hello @AcidCool83 , do you confirm that you screen is directly connected to the HDMI port of your raspberry pi?

anxuae, 2021-01-31:
Hi @AcidCool83 

I have no news about your issue, is it fixed?
If yes, can you close the issue?

Thanks
anxuae

hutattedonmyarm, 2021-02-05:
I'm having the same issue. Using the Pi HQ camera, and the official 7" Pi touchscreen

Logs are:

`pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: neopixelflash-0.0.1
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 2B
error: XDG_RUNTIME_DIR not set in the environment.
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ INFO    ] pibooth           : Hello from 'pibooth_neopixelflash' plugin. 8 RGBW LEDS @ pin: 21
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : BUTTONDOWN: generate CAPTURE button event
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : +< Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : +> took 0.570 seconds
[ DEBUG   ] pibooth           : BUTTONDOWN: generate CAPTURE button event
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : +< Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[... 86 lines cut ...]
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to concatenate images
[ DEBUG   ] pibooth           : |  +> took 2.143 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to assemble final image
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to draw texts
[ DEBUG   ] pibooth           : |  +> took 0.000 seconds
[ INFO    ] pibooth           : |  +< Use PilPictureFactory to outline boundary borders
[ DEBUG   ] pibooth           : |  +> took 0.004 seconds
[ DEBUG   ] pibooth           : +> took 2.313 seconds
[ INFO    ] pibooth           : +< Save image '/root/Pictures/pibooth/2021-02-05-19-12-13_pibooth.jpg'
[ DEBUG   ] pibooth           : +> took 0.868 seconds
[ DEBUG   ] pibooth           : Activate state 'finish'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] pibooth           : Add to buffer the image '1802248784'`

anxuae, 2021-05-05:
Hi @hutattedonmyarm  and @AcidCool83,

Did you find a way to solve your issue?

BR

hutattedonmyarm, 2021-05-05:
Sadly no

anxuae, 2021-05-05:
I have not HQ camera, so I can not test by myself.
Can you confirm that your screen is connected using the HDMI port of the raspberry?
What is the resolution of your screen, what resolution do you have set in the configuration?

Here below is a small script to test the preview and capture of the picamera, can you write it in a file named `test.py`

```python
import picamera
import time

camera = picamera.PiCamera()
camera.resolution = (1920, 1080)
camera.framerate = 15
camera.start_preview()
camera.annotate_text = 'Hello world!'
time.sleep(10)
# Take a picture including the annotation
camera.capture('foo.jpg')
camera.stop_preview()
```

then run it in a shell:

```
python test.py
```

hutattedonmyarm, 2021-05-06:
> Can you confirm that your screen is connected using the HDMI port of the raspberry?
> What is the resolution of your screen, what resolution do you have set in the configuration?

I'm connecting the official touch screen via their display interface, so 800 x 480 resolution. Config is set to `fullscreen`.

I think I have found the issue!
So far, I've been ssh-ing into the pi and starting pibooth from my computer. Doing this, everything works, except the preview. If I VNC in and start pibooth from there, the preview works, even after closing VNC again.
Setting pibooth to autostart solves this too

anxuae, 2021-05-06:
Yes you are right: the preview is an overlay managed at GPU low level, thus it is not accessible using ssh (which use X11 forwarding, a hight level HMI management)
`pibooth` has to be started from a shell on the raspberry-pi (HDMI screen, or display interface screen).

I think that it is the same issue for @AcidCool83

anxuae, 2021-05-06:
Add warning in README: https://github.com/pibooth/pibooth/commit/b5466e3c085c762da4f9a6c23cbb6e608493649b

Please @AcidCool83 reopen the issue if your problem is different and not solved.

BR
anxuae

closed 2021-05-06 by anxuae