pibooth/pibooth #188: Tactile setup

TheSuperpapi, 2020-10-11:
Hello there,

Here is what was updated to ptopose this tactile interface.

[Click on me](https://youtu.be/FaiWML-OOus)

In the background.py (just replace the .png file by what you want):

`# -*- coding: utf-8 -*-

import os.path as osp
import pygame

from pibooth import fonts, pictures
from pibooth.language import get_translated_text

ARROW_TOP = 'top'
ARROW_BOTTOM = 'bottom'
ARROW_HIDDEN = 'hidden'

def multiline_text_to_surfaces(text, color, rect, align='center'):
    """Return a list of surfaces corresponding to each line of the text.
    The surfaces are next to each others in order to fit the given rect.

    The ``align`` parameter can be one of:
       * top-left
       * top-center
       * top-right
       * center-left
       * center
       * center-right
       * bottom-left
       * bottom-center
       * bottom-right
    """
    surfaces = []
    lines = text.splitlines()

    font = fonts.get_pygame_font(max(lines, key=len), fonts.CURRENT,
                                 rect.width, rect.height / len(lines))
    for i, line in enumerate(lines):
        surface = font.render(line, True, color)

        if align.endswith('left'):
            x = rect.left
        elif align.endswith('center'):
            x = rect.centerx - surface.get_rect().width / 2
        elif align.endswith('right'):
            x = rect.right - surface.get_rect().width / 2
        else:
            raise ValueError("Invalid horizontal alignment '{}'".format(align))

        height = surface.get_rect().height
        if align.startswith('top'):
            y = rect.top + i * height
        elif align.startswith('center'):
            y = rect.centery - len(lines) * height / 2 + i * height
        elif align.startswith('bottom'):
            y = rect.bottom - (len(lines) - i) * height
        else:
            raise ValueError("Invalid vertical alignment '{}'".format(align))

        surfaces.append((surface, surface.get_rect(x=x, y=y)))
    return surfaces

class Background(object):

    def __init__(self, image_name, color=(0, 0, 0), text_color=(255, 255, 255)):
        self._rect = None
        self._outlines = True
[... 464 lines cut ...]

    def __init__(self):
        Background.__init__(self, "finished")

    def resize_texts(self):
        """Update text surfaces.
        """
        rect = pygame.Rect(self._rect.width * 0.3, self._rect.height * 0.6 - self._text_border,
                           self._rect.width * 0.4, self._rect.height * 0.4)
        Background.resize_texts(self, rect)

class OopsBackground(Background):

    def __init__(self):
        Background.__init__(self, "oops")

`

Globaly, as you can see, I only replaced the arrow PNG by other PNG and changed size and position.
If you need more, please ask.

Sorry for the delay and this post quality, but without internet I spent 1H to make and post it.... hope my internet issue will be solved soon.

Best Regards to all of you.
Papi

werdeil, 2020-10-12:
Thanks a lot @TheSuperpapi we will see how to integrate your work on a Pull request.

sravel, 2020-10-14:
yes great job @TheSuperpapi . I would also like to test because I find it nice to have the tactile side, is it possible to share your images from the video?

TheSuperpapi, 2020-10-14:
Hello, here are the tactil .png icon,

Let me know if you can open and use this zip file.

You must copy and past it in the right folder.

[Tactil png icon.zip](https://github.com/pibooth/pibooth/files/5377167/Tactil.png.icon.zip)

See you

werdeil, 2020-10-14:
I have pushed a PR to have a touchscreen option in #191. I've used other icon to be sure to credit the owners.

Here is an example of how it renders: https://youtu.be/PpSD7dVkaiM

anxuae, 2020-11-14:
Hi @TheSuperpapi ,

The PR is now merged, the feature will be available in pibooth 2.0.2

Thanks for reporting.
anxuae

closed 2020-11-14 by anxuae