pibooth/pibooth #190: qr code

gliepins, 2020-10-13:
Dear friends,
Do you have any working solution to generate a QR code for picture download? 
Thanks!

werdeil, 2020-10-14:
You can have a look at the qrcode plugin: https://github.com/pibooth/pibooth-qrcode however this plugin is not yet finished.

TheSuperpapi, 2020-10-14:
On my side, i'm generatin QR code of the google, photo album sharing link via website like this one:

[Site](https://fr.qr-code-generator.com/solutions/app-qr-code/?ut_source=google_c&ut_medium=cpc&ut_campaign=franzoesisch_dsa&ut_content=solutions&ut_term=_b&gclid=CjwKCAjww5r8BRB6EiwArcckC09BZwxsDRqSlowVIgicH10Yiyx_nwco4A1VrmGxJT_BcP-HO3G7zRoCGLYQAvD_BwE)

And I print it and put the paper in a picture frame.

Hope it can help you if the qrcode plugin doesn't respond to your need

See you.

gliepins, 2020-10-14:
> You can have a look at the qrcode plugin: https://github.com/pibooth/pibooth-qrcode however this plugin is not yet finished.
Thanks guys!
The plugin you mentioned - is it even working?? Has anyone updated it to be usable?

gliepins, 2020-10-14:
> On my side, i'm generatin QR code of the google, photo album sharing link via website like this one: [...]

Thank you!
The qrcode generation itself is not a problem. qrencode can do that perfectly. What I did, is I take the photo generated by pibooth put it up on instant webaccess right there on the Raspberry. Except that I would want the qrcode displayed right there on the photobooth screen. But for that I need to dig into the pibooth code.
One way I see this is working is this: There are "assets" inside the files of the code. And I could replace dynamically one of the assets on screen with the newly generated qrcode. And once that is working the link inside the qrcode can be whatever I set it to be - a google link (if the photo is uploaded there), or a local http address to download the photo there on the spot. Of course local wifi network should be accessable to do that.
That is why I was asking about the QR plugin. Can anyone tell how that plugin was supposed to operate? Maybe there is a celaner way to display the QR code on the screen?

werdeil, 2020-10-14:
Currently the plugin is displaying a qrcode at the wait screen (main screen), I guess it what you want so no need to go deep in the pibooth code.

You can open an issue on the plugin repo and we will help you from there.

gliepins, 2020-10-14:
> Currently the plugin is displaying a qrcode at the wait screen (main screen), I guess it what you want so no need to go deep in the pibooth code.
> 
> You can open an issue on the plugin repo and we will help you from there.

Thanks, I will get back with the update!

sravel, 2020-10-14:
I had made a version which take a fixed link or add a base URL to the name of the generated photo, I don't know if that can answer your problem
@werdeil I agree to use my code to append on the official release.
https://github.com/sravel/pibooth-qrcode

gliepins, 2020-10-14:
Thanks!!! I will test that! You all have been short and very helpful! Thats means great!! 👍 
p.s  I'm also thinking on how to bring this all project away from Raspberry over to Arduino or similar to that hardware!

anxuae, 2020-11-01:
Hi everybody.
I close this issue, because there is now an official plugin to support this feature: https://github.com/pibooth/pibooth-qrcode

closed 2020-11-01 by anxuae

gliepins, 2021-02-05:
Guys, Happy New Year!
I cannot get around the qrcode. 
I upgraded the pibooth to latest,
I installed the plugin sucessfuly,
now running the pibooth app it throws an error! 
Please, help!
[image]
[image]