pibooth/pibooth #192: Automatic run a script with pibooth, NOT a plugin

DJ-Dingo, 2020-10-15:
I have a PIR sensor script that I run before i open Pibooth. 
Is there a way to run it automatic when pibooth start, just like plugins?

Since it has some time.sleep() in it, i can´t run it as a plugin, it will have to be rewriting to work as a plugin.

werdeil, 2020-10-15:
An ugly way is to make an alias to pibooth that runs both pibooth and your script. But I find it very ugly.

As you gave me access to your PIR sensor code I'll try to find a way to turn it into a plugin, I think this is possible.

DJ-Dingo, 2020-10-15:
Thx @werdeil  Yes, it is :-), 

It only need to be active in state_wait.
And proberly a better way to handle sounds from a folder instead of direct link to each file.

The script was something i made for an arcade machine, but has been working nice in the background with pibooth aswell.

closed 2020-10-15 by DJ-Dingo