pibooth/pibooth #194: Countdown is showing white transparent rectangles

mmk8283, 2020-10-25:
When using a webcam the countdown is displayed as white transparent rectangles, if using a pi camera the countdown appears correctly. I have tried different versions of Pillow with no change in the result.

werdeil, 2020-10-26:
Can you gave us your python packages configuration?

pip freeze > pip_configuration.txt

and paste it here?

anxuae, 2020-11-14:
I @mmk8283 ,

Any news on this topic?

Thanks
anxuae

sravel, 2020-11-18:
[pip_configuration.txt](https://github.com/pibooth/pibooth/files/5558678/pip_configuration.txt)
[pibooth.log](https://github.com/pibooth/pibooth/files/5558700/pibooth.log)

I @anxuae 
I update to last master version and I have the same issue on my dell XPS,
I also need to install openCV to see integrated camera (but no need on old version?)
[pibooth.log](https://github.com/pibooth/pibooth/files/5558700/pibooth.log)

werdeil, 2020-11-18:
About the white rectangle can you downgrade your `pillow` version to 7.1.2? 

`pip install pillow=7.1.2`

About the integrated camera, are you sure you didn't install opencv on the old version? I don't see how it could have worked without it.

sravel, 2020-11-18:
For openCV i found the reason, I just updated to latest Ubuntu 20.04 version so default python change without modules!
It's worked with this version of pillow thanks.
pip install pillow==7.1.2

detrist, 2021-03-05:
The same problem appeared on RPI 4B, resolved by increasing the version of pillow to 7.1.2.

anxuae, 2021-04-23:
Upgraded Pillow to 7.1.2 mini, see: https://github.com/pibooth/pibooth/commit/c89b3e6d59bbc915e1976ce7baadb58be94ebdd0

closed 2021-04-23 by anxuae