pibooth/pibooth #195: Can´t use % in footer text

DJ-Dingo, 2020-10-27:
If you use % in the footer text, pibooth fails

werdeil, 2020-10-27:
Good catch, indeed I got a crash too, I'll have a look

DJ-Dingo, 2020-10-27:
If you use %% it works. but not %

werdeil, 2020-10-27:
The issue comes from the fact the ConfigParser interpolate the % in the value while setting the config.

I see 2 solutions:
- Inherit from RawConfigParser instead of ConfigParser
- Escape automatically the "%" into "%%" just before setting the value in the config.

@anxuae which solution do you prefer? Do you see a reason to not use RawConfigParser (interpolation used elsewhere?)

anxuae, 2020-11-01:
Yes RawConfigParser can be used

werdeil, 2020-11-03:
@DJ-Dingo I just pushed a fix in the master branch, it shall fix your issue too (on pibooth-lcd plugins) reopen the issue if it is not the case.

commit 87a1d48d (2020-11-03): Use RawConfigParser instead of ConfigParser to allow "%" in the config (fix #195)
closed 2020-11-03 by werdeil