pibooth/pibooth #196: Rotate Picture

sonkinshu, 2020-10-29:
I need to invert the picture but its ignoring the changes I've made to the config file. Attempting to rotate 180.

DJ-Dingo, 2020-10-29:
@sonkinshu I just tried 180 and it's working fine here - 180
with version 2.0.1 and raspberry pi cam

Remember to NOT have the pibooth program running when you make changes in the config.cfg
First Exit pibooth, Then open the config.cfg, Make changes and save and exit config cfg again. Then open pibooth again.

werdeil, 2020-11-03:
@sonkinshu Did you managed to make it work? which setting in the config did you changed?

anxuae, 2020-11-14:
Hi @sonkinshu ,

Have you solved your issue? If yes, can you close it?

Thanks

closed 2020-11-21 by sonkinshu

dafunkychild, 2021-02-17:
Hello, I did follow the instructions but modifying rotation parameter in the config file as no effect.