pibooth/pibooth #198: Overlay Image - black areas of overlay become transparent

mmk8283, 2020-10-29:
I'm noticing that an overlay containing pure black (0,0,0) becomes transparent or acts as alpha in those areas. has anyone else experienced this?  maybe I'm doing something wrong...

DJ-Dingo, 2020-10-29:
@mmk8283 
Verry Nice Halloween Overlay, something you want to share?

mmk8283, 2020-10-29:
thank you, 

I will gladly share it, however, areas turn transparent, and it doesn't work effectively work as an overlay. I just modified the file and lightened the whole image so that there are no areas of (0, 0, 0) in the image. I will see if that solves the issue tonight.

DJ-Dingo, 2020-10-29:
Okay let me know if it works :-) Do you have it as a pibooth-picture-template file

mmk8283, 2020-10-29:
no sorry, I only made the one for the 2 picture portrait style. kind of a spur of the moment project

DJ-Dingo, 2020-10-29:
okay, but if you have it working and have it in another file, could be fun to play around with it in the picture-template.
I am not gonna use it this year due to corona. So i had plenty of time to have it ready for next year :-)

werdeil, 2020-11-01:
@mmk8283 I'm not sure to understand what is going on, can you provide us an example on how it rendered?

mmk8283, 2020-11-02:
[image]

DJ-Dingo, 2020-11-02:
Hey @mmk8283 I just testet your overlay, and i did not have the same issue as you did.
( here with a quick edit to 1 pic "not perfect made, but just a test" ) 

[image]
[image]

werdeil, 2020-11-03:
Ok I reproduced the issue, I'll try to find a solution

werdeil, 2020-11-03:
Found a fix, it was only on cv2 factory not oil factory so if you don't have cv2 installed the issue wasn't there (I guess this was the case for @DJ-Dingo.

closed 2020-11-03 by werdeil

mmk8283, 2020-11-03:
> Found a fix, it was only on cv2 factory not oil factory so if you don't have cv2 installed the issue wasn't there (I guess this was the case for @DJ-Dingo.

Thanks so much for your help! I will look into my setup and make the change if necessary.

srvoets, 2022-10-28:
sorry to resurrect this but is there documentation anywhere for overlay creation?  or a repo of overlays others hve made somewhere?

mmk8283, 2022-10-28:
Hi again, I looked back at the pictures from our halloween party two years ago, and it looks like I never got it resolved... I ran out of time and just went with the transparency issue. I never got around to trying to remove CV2 factory... and honestly I still don't know how to. I will dig around to try to find the old templates I used... I'm not sure I have them on this computer.

mmk8283, 2022-10-28:
This is what I could find, I made an attempt to select everything that was 0,0,0 causing the transparency issue and changed it to a dark gray... I cannot confirm if this works, I'm not positive I was able to test it. 

In more recent use of my photobooth, I have been making nice custom backgrounds and using the three diagonal pictures instead.... seems to produce results that everyone is happy with.  And as of recently my monitor I've been using for the photobooth died... so its a project on the back burner for now. Good luck with your project!

[image]
[image]
[image]

DJ-Dingo, 2022-10-29:
Nice @mmk8283 thanks. I will check them out this weekend. If you happen to find the original files so we could remove 2020 without having to edit too much that would be cool too.

@werdeil as @srvoets mentions, it could be cool to collect other people's overlays to share. Maybe make a permanent thread or a folder, with both pictures and picture template xml files.

@srvoets In principle, it is just a picture frame with a transparent where the original picture must be visible. So you can use many of them that are available for free download online. Try search "halloween frame transparency" on google images, and you get alot. However, in some/many cases you will have to make your own edit/pictures, and/or use "pibooth-picture-template" plugin to make it work/fit with text and size in pibooth - https://github.com/pibooth/pibooth-picture-template

Personally, I mostly use 1 image that stands up.. So many of my frames only work with this setup. In principle, I should do as @mmk8283 has done and make it so that it works for 1,2,4 taken photos in the same frame.

commit 40eb7261 (2020-11-03): Rework cv2 factory to handle correctly RGBA overlays (fix #198)