pibooth/pibooth #199: new install - Import Error

MicroMast3r, 2020-10-30:
Hello,

i installed the program according to the instructions, but i can't get it to work. when I enter "pibooth" in the console, the following comes up:
pi@raspberrypi:~ $ pibooth
pygame 2.0.0 (SDL 2.0.9, python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/home/pi/.local/bin/pibooth", line 6, in <module>
    from pibooth.booth import main
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/booth.py", line 28, in <module>
    from pibooth.config import PiConfigParser, PiConfigMenu
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/config/__init__.py", line 4, in <module>
    from pibooth.config.menu import PiConfigMenu
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/config/menu.py", line 7, in <module>
    import pygame_menu as pgm
  File "/home/pi/.local/lib/python3.7/site-packages/pygame_menu/__init__.py", line 51, in <module>
    import pygame_menu.font
  File "/home/pi/.local/lib/python3.7/site-packages/pygame_menu/font.py", line 34, in <module>
    import pygame.font as _font
ImportError: libSDL2_ttf-2.0.so.0: cannot open shared object file: No such file or directory

it's because the python version is too high?
if so how can i downgrade this?

werdeil, 2020-10-30:
No need to change your python version.

The issue comes from the fact that `pygame`just released the version 2.0.0 of their package ([https://github.com/pygame/pygame/releases](https://github.com/pygame/pygame/releases)) and it raised issues with the package `pygame-menu`

A new version of `pygame-menu` have been released too to make it work with it: [https://github.com/ppizarror/pygame-menu/releases/tag/3.3.0](https://github.com/ppizarror/pygame-menu/releases/tag/3.3.0)

Can you check that both pip packages are up to date?

MicroMast3r, 2020-10-30:
so according to "pip list" is 
pygame version 1.9.4.post1

and according to "pip3 list" is 
pygame 2.0.0
pygame-menu 3.3.0
pibooth 2.0.1

werdeil, 2020-10-30:
Indeed with this configuration I have the same issue. we'll work with Pygame-menu team to correct it, in the meantime you can downgrade your pygame version to 1.9.6: 
`pip3 install pygame==1.9.6`

MicroMast3r, 2020-10-30:
when I try to start now comes 

pi@raspberrypi:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 6, in <module>
    from pibooth.booth import main
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/booth.py", line 29, in <module>
    from pibooth.config import PiConfigParser, PiConfigMenu
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/config/__init__.py", line 4, in <module>
    from pibooth.config.menu import PiConfigMenu
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/config/menu.py", line 8, in <module>
    import pygame_vkeyboard as vkb
ModuleNotFoundError: No module named 'pygame_vkeyboard'

do I also do other things wrong? :/

MicroMast3r, 2020-10-30:
okay i downgrade pygame-menu to 3.1.0 and Pillow to 6.1.0
now it looks like it's about to start:

pi@raspberrypi:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
[ INFO    ] pibooth           : Configuring Picamera camera ...
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 456, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 112, in __init__
    config.getboolean('CAMERA', 'delete_internal_memory'))
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/__init__.py", line 39, in get_camera
    return cam_class(iso, resolution, rotation, flip, delete_internal_memory)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/rpi.py", line 55, in __init__
    self._cam.resolution = resolution
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 2271, in _set_resolution
    "Invalid resolution requested: %r" % (value,))
picamera.exc.PiCameraValueError: Invalid resolution requested: PiResolution(width=1934, height=2464)

do you have an idea for this Problem?

werdeil, 2020-10-30:
For this issue you need to use a resolution supported by your camera, have look at the following link to see which one are correct (the link is down on my side but should work soon):
http://picamera.readthedocs.io/en/latest/fov.html#sensor-modes

Which version of the camera do you use?

MicroMast3r, 2020-10-30:
When I bought the camera it was said to be a raspberry with 5 Mpix and zoom. unfortunately i don't have more information. =/

MicroMast3r, 2020-10-30:
it is a new issue?
[pibooth.log](https://github.com/pibooth/pibooth/files/5465922/pibooth.log)

or is it the same - camera error?

at least I can start the program.

werdeil, 2020-10-30:
Linked to the same issue, you put the resolution in the wrong format, resolution shall respect this format: (1934, 2464)

There is an issue with the format you provided.

closed 2020-10-31 by MicroMast3r