pibooth/pibooth #200: Connection to Cups

MicroMast3r, 2020-10-31:
Hello,

I have a problem with the connection to cups and don't know how to get this away.

pi@raspberrypi:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
[ INFO    ] pibooth           : Configuring Picamera camera ...
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/booth.py", line 456, in main
    app = PiApplication(config, plugin_manager)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/booth.py", line 126, in __init__
    self.count)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/printer.py", line 36, in __init__
    self._conn = cups.Connection() if cups else None
AttributeError: module 'cups' has no attribute 'Connection'

i can see the printer in cups and also print test pages.
I want to use a Canon Selphy CP1300

anxuae, 2020-11-01:
Can you run this command:

`pip freeze | grep pycups`

And send me the result?

anxuae, 2020-11-14:
Hi @MicroMast3r 

What is the status of this bug? Do you solve it?

If yes, can I close this issue?

thanks

anxuae, 2021-04-23:
No feedback since several months, I'm closing this issue.
Please re-open if necessary.

closed 2021-04-23 by anxuae