pibooth/pibooth #201: Touchscreen action not taken into account with pygame 2.0.0

werdeil, 2020-11-02:
**Describe the bug**
Install pibooth with pygame 2.0.0 (newly released)
Run it on a raspberry pi with touchscreen, touch action are not taken into action.

Note: everything works as expected if we revert pygame to the 1.9.6 version

**Logs**
Touch action gives no logs in debug mode.

**Environment**
 - OS : Raspbian Buster
 - Camera Rpi camera v2
 - Pibooth version 2.0.1
 - Official Raspberry Pi touchscreen 7"

retromarcia, 2020-11-18:
Hey @werdeil,

at the moment a possible solution to work with a 5" touchscreen it's to downgrade the pygame version? I have noticed another thing I don't know if it correlates to this problem...using pi camera v.2.1 and a capacitive touchscreen display the image taken seems darker or with weird color...if a connect to the normal LCD everything works fine...

werdeil, 2020-11-19:
@retromarcia Yes you can downgrade pygame to version 1.9.6 until we release a new version with touchscreen support.

About the darker and weird color issue, if you think it is related to pibooth you can open a new issue and provide screenshots/photos.

retromarcia, 2020-11-19:
Perfect! Thanks @werdeil  :)

romaingerardy, 2021-01-08:
@werdeil When i downgrade the 2.0.1 realease (tags) with pygame 1.9.6 to be able to use the touchscreen, i got this error:

  File "/home/pi/pibooth/pibooth/view/window.py", line 38, in __init__
    info = pygame.display.Info()
pygame.error: video system not initialized

Did you fix this ? 

Thanks !

werdeil, 2021-01-08:
The problem appears if pygame doesn't find a X server, did it happen when you try to launch pibooth through ssh?

romaingerardy, 2021-01-08:
Well i ""fix"" my issue by hardcoding the window size... Then i tried to restart pibooth and another problem appeared: it was because i tried to launch pibooth through ssh (with pygame 2 it was ok). So i launched it directly on the raspberry. 

Now it's ok, but unfortunately the mouse coordinates are incorrect. It seems that the mouse is not calibrated with the touch. Don't know if anyone encountered the problem ?!

closed 2021-05-07 by anxuae