pibooth/pibooth #203: New Plugin

ceeeeb, 2020-11-04:
Hello,

I created an plugin to send the photos to nextcloud and you can access it afterwards via a QR code.

Are you interested in this plugin?

werdeil, 2020-11-04:
Yes we are, can you give the link of the plugin here? We will have a look

ceeeeb, 2020-11-04:
herefind the github link
https://github.com/ceeeeb/pibooth-nextcloud

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

anxuae, 2021-05-10:
Hi @ceeeeb,

Thank for sharing your plugin and great job! It will be displayed when the visitor click on the link in the README section [Customize using plugins](https://github.com/pibooth/pibooth#customize-using-plugins).

However to integrate and move your repository in the `Pibooth` organisation (of course welcomed 😄), you will need to give me all the admin rights on you repository. So it is really up-to..

BR
anxuae

ceeeeb, 2021-05-11:
Hi anxuae,

Can you give me your email address so I can give you the administrator
access

anxuae, 2021-05-22:
Hi @ceeeeb 

I let you do the things if you want. But this issue can be closed

closed 2021-05-22 by anxuae

DJ-Dingo, 2021-05-22:
@anxuae 

Curious, I can not see any info about this plugin, in the "link in the README section".
Did i look the wrong place, or you forget to insert it ?

mozdi, 2023-02-19:
Hi @ceeeeb, @anxuae,
is this plugin still working with pibooth?
I couldn´t find it in the README section or on any other place where plugins are linked.

I´ve raised an issue in the plugin, because the installation using pip fails:
https://github.com/ceeeeb/pibooth-nextcloud/issues/1

BR Mozdi