pibooth/pibooth #204: Opening the menu while Picamera preview is on doesn't close the preview

werdeil, 2020-11-12:
**Describe the bug**
As described in the title, if you open the menu while the preview is on, the menu opens and the preview stays on (on top of the menu).

2 solutions for me:
- Do not allow menu opening during the preview and capture part
or
- Force preview stop when the menu shows

@anxuae tell me which one you prefer

**Environment**
 - OS: Raspbian Buster
 - Camera : Rpi camera
 - Pibooth version 2.0.1

anxuae, 2020-11-14:
Hello @werdeil 

Both are ok. Maybe it is more user-friendly to let opening the menu, it permit to stop the current sequence during tests for instance...

closed 2020-11-16 by werdeil

werdeil, 2020-11-16:
It was one line of code so I didn't made a PR.

commit 300817e0 (2020-11-16): Stop camera preview when entering setting menu (fix #204)