pibooth/pibooth #205: Temporarily turn off installed plugins

DJ-Dingo, 2020-11-13:
Is there a way to temporarily turn off an installed plugin, without losing the info in the config.

Sometimes I need to temporarily turn off a plugin, especially when testing other plugins I am working on. But also if in some cases they are not to be used for that evening/event.

werdeil, 2020-11-13:
This feature is not yet present but I'll try to find a good solution for this.

DJ-Dingo, 2020-11-13:
Maybe an Admin menu with an code, with other functions that normal users should not have access to, also can be changed or turned on / off.

[Like i mentions here](https://github.com/pibooth/pibooth/issues/42#issuecomment-707953982)

anxuae, 2020-11-14:
Hi guys,

@werdeil  I can work on this topic if you want.
I propose a dynamic menu to enable/disable installed plugins

anxuae, 2021-04-28:
Implemented in PR #230

closed 2021-04-28 by anxuae