pibooth/pibooth #207: pibooth is not starting

qchatenet, 2020-11-19:
Hi,

When I try to start pibooth with command `$ pibooth `, I get the following error : ImportError: libSDL2_ttf-2.0.so.0: cannot open shared object file: No such file or directory

I followed the installation instructions on a fresh Raspbian OS installed on a raspberry 3 board (tried it several times, with no success).

Thank you for your help.

werdeil, 2020-11-19:
This comes from the new version of pygame (version 2.0.0).

Two options available:
- Install pygame version 1.9.6 (`pip install pygame==1.9.6`)
or
- Keep your pygame version and follow the following instructions:
https://www.piwheels.org/project/pygame/
More precisely the apt line:
`sudo apt install libasyncns0 libflac8 libfluidsynth1 libice6 libjack0 libjbig0 libmodplug1 libmpg123-0 libogg0 libopus0 libopusfile0 libportmidi0 libpulse0 libsdl2-2.0-0 libsdl2-image-2.0-0 libsdl2-mixer-2.0-0 libsdl2-ttf-2.0-0 libsm6 libsndfile1 libsndio7.0 libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libwayland-client0 libwayland-cursor0 libwayland-egl1 libwebp6 libx11-xcb1 libxcursor1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2 libxrender1 libxss1 libxtst6 libxxf86vm1`

qchatenet, 2020-11-19:
Hi @werdeil , thank you for the help, it is perfectly running now. Congratulations for pibooth, it is an awsome project !

closed 2020-11-19 by qchatenet