pibooth/pibooth #208: How to show the correct image at chosen_enter.

DJ-Dingo, 2020-11-20:
Is there a way to display the correct images at chosen_enter.
I am working on a plugin that can show a image on an small OLED display on each state.
But how can i display the correct chosen image?

Now i have it to just show the same image layout1.png at chosen_enter.
But if people choose layout2, layout3 or layout4, it will be cool if it will show the correct image layout(1,2,3,4).png  on the display. Just like it does on the pibooth screen.

@pibooth.hookimpl
def state_chosen_enter(app):
    # Write state picture on OLED screen at chosen_enter
    try:
        s = app.states_pictures.split()
        if "Yes" in s:
            if app.device.height == 32:
                app.image = Image.open('/home/pi/.config/pibooth/oled_states/layout1_32.png').convert('1')
            elif app.device.height == 64:
                app.image = Image.open('/home/pi/.config/pibooth/oled_states/layout1_64.png').convert('1')
        # Display image
        app.device.display(app.image)
    except OSError:
        pass

werdeil, 2020-11-20:
I see two part of solution: first the selected info is present in `app.capture_nbr`so you shall have access to it in the plugin.

Secondly I can't test but I guess you can use directly the `get_pygame_layout_image` method from the `picture` part:

```
from pibooth.pictures import get_pygame_layout_image

@pibooth.hookimpl
def state_chosen_enter(app):
    # Write state picture on OLED screen at chosen_enter
    try:
        s = app.states_pictures.split()
        if "Yes" in s:
            layout_image = get_pygame_layout_image(app.window.text_color, app.window.color, app.capture_nbr, (app.device.height, app.device.height))
        # Display image
        app.device.display(layout_image)
    except OSError:
        pass
```

DJ-Dingo, 2020-11-20:
okay :-) 

But i have to use my own small 128x32 or 128x64 8bit images. (layout1_32.png or layout1_64.png)
So guess i need to get info from the app.capture_nbr, to deside what of my own image to show on the screen.

My images is etc. 
On an 128x32 oled-screen:
layout1_32.png or  layout2_32.png or layout3_32.png or layout4_32.png
On an 128x64 oled-screen.
layout1_64.png or  layout2_64.png or layout3_64.png or layout4_64.png

werdeil, 2020-11-20:
Yes I guess that you can't use directly the pygame image, so use something like that to get directly the correct image:

```
@pibooth.hookimpl
def state_chosen_enter(app):
    # Write state picture on OLED screen at chosen_enter
    try:
        s = app.states_pictures.split()
        if "Yes" in s:
            app.image = Image.open('/home/pi/.config/pibooth/oled_states/layout{0}_{1}.png'.format(app.capture_nbr, app.device.height)).convert('1')
        # Display image
        app.device.display(app.image)
    except OSError:
        pass
```

DJ-Dingo, 2020-11-20:
Cool, Working great on my 128x32 
Thanks!!

I dont have a printer, but Is there a specific command in use if printer fails?
Or is it just using the normal state_failsafe. images in :
@pibooth.hookimpl
def state_failsafe_enter(app):

werdeil, 2020-11-23:
The printer part is handled by CUPS, pibooth will only get the status of the printer (if it is available or not) et send the picture to print. There shall not be specific failure on this part as CUPS failure won't affect pibooth.

DJ-Dingo, 2020-11-23:
Ok thx.
That was just because I saw printer_failure.png in the image folder.

closed 2020-11-23 by DJ-Dingo