pibooth/pibooth #209: Force "0" if user does not enter a number in Pibooth Menu option

DJ-Dingo, 2020-11-30:
Is it possible to force "0" in config.cfg if user does not enter an number or accidentally leave field empty.

cfg.add_option('OLED DISPLAY TEXT', 'oled_text3_right', "0",
               'Text-3 move right',
               "Text-3 move right", "0")

werdeil, 2020-12-23:
No there is not possible for now, one solution would be to put a default value when you get the value in your plugin. It won't change the saved value but it will allow to work in a default mode.

DJ-Dingo, 2020-12-23:
OK, thanks. It's more because in my new OLED plugin if you forget to enter something, then the OLED screen will just be black.

So in future versions of pibooth if there is a way to double check for that or inform the user that this field should not be empty, it could be usable. Like some code that would make the field "red" or something.

closed 2020-12-23 by DJ-Dingo