pibooth/pibooth #210: Show Mouse Cursor in Fullscreen - Settings Option

Groess, 2020-12-09:
If someone has no buttons attached and no touchscreen it would be nice to see the mouse cursor to easily click to take a picture.  

An option in the config file or in the settings menu to switch between "show or hide mouse cursor in fullscreen mode"

DJ-Dingo, 2020-12-15:
Maybe an option to use both the buttons on the mouse as the 2 buttons for pibooth when you are in full screen only.

Groess, 2020-12-15:
That would be also great!

werdeil, 2020-12-15:
> Maybe an option to use both the buttons on the mouse as the 2 buttons for pibooth when you are in full screen only.

Yeah I've thought of that but this is quite difficult as on pygame2 touch action are seen as mouse clicks and there is no mean to know if there is a touch screen or not.

We could add an option to allow the display of cursor even in fullscreen but I'm not a fan of adding more and more options that are only used on one setup. 

I can't test it but instead of putting the app in fullscreen maybe you can put the resolution as the one of you screen. As it is not fullscreen you shall keep the cursor and still have pibooth on the whole screen.

DJ-Dingo, 2020-12-16:
> I can't test it but instead of putting the app in fullscreen maybe you can put the resolution as the one of you screen. As it is not fullscreen you shall keep the cursor and still have pibooth on the whole screen.

@werdeil 
That is working just fine. I been doing that when testing plugins.

So @Groess you just have your setup in pibooth settings / Window / Startup size to **not** be fullscreen, set it to etc. 800x400. 

When you open Pibooth you just **press** the fullscreen option only for the pibooth window  **the middle option in the up-right corner of that window**

Now you have your mouse on all the time.

To make it look more like fullscreen, make it **hide your panel** on the screen.
Go into **panel setup** in the raspberry pi (just right click at the panel-bar). And under advanced Tap and activate the **auto hide panel**. You can also set the size of the minimized panel (mine is set to 2). When you need to use you panel you can then just go over the hide panel with the mouse and it will popup. (just make sure that the Pibooth cmd-window is **not** in fullscreen when you do that. Just click at the same option again for the pibooth window  **the middle option in the up-right corner of that window**
).

Think this is the best solution to what you want to do.

Groess, 2020-12-16:
Thanks a Lot!!!

DJ-Dingo, 2020-12-16:
> Thanks a Lot!!!

Your welcome :-)

DJ-Dingo, 2020-12-16:
And @Groess you can also Hiding the window decoration bar for the maximized window.
Just right-click on the gray bar in the top of the pibooth CMD-window when you have it in fullscreen and remove that one. 

I am not using english on my Pi but think it is called add / remove - decoration  (The last option in the popup window). Now you have a fullscreen version, just like fuldscreen in pibooth opstions, but with mouse.

It can be a little tricky to use, becouse you now don´t have a gray bar to maximize and minimize. So if you done this when it was minimized, you need to maximize it through the panel-bar. You can do that in the pop-up panel. Right click the Pibooth-CMD and mazimize it. After that just go all the way to the top of the screen again and add the gray bar again when you need it. 

Maybe you can also use "**ALT + Tap**" to switch between them, haven´t testet it.

closed 2020-12-16 by Groess