pibooth/pibooth #212: Pibooth crash

arthur5640, 2021-01-06:
When i put twice on escape, i have a crash and i can't execute pibooth. Can you help me 

Labels
bug
Comments
@arthur5640 arthur5640 commented 15 minutes ago
hello, when i put on escape twice , i have the crash and i can't execute pibooth.
How i can resolve this:

**

                Oops! It seems that pibooth has crached                     *
You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
                  and post the file: /tmp/pibooth.log                       *
pi@raspberrypi:~ $ pibooth
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
File "/usr/local/bin/pibooth", line 10, in
sys.exit(main())
File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 425, in main
config = PiConfigParser("/.config/pibooth/pibooth.cfg", plugin_manager)
File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 256, in init
self.load()
File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 294, in load
self.handle_autostart()
File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 308, in handle_autostart
enable = self.getboolean('GENERAL', 'autostart')
File "/usr/lib/python3.7/configparser.py", line 828, in getboolean
raw=raw, vars=vars, fallback=fallback, **kwargs)
File "/usr/lib/python3.7/configparser.py", line 808, in _get_conv
**kwargs)
File "/usr/lib/python3.7/configparser.py", line 802, in _get
return conv(self.get(section, option, **kwargs))
File "/usr/lib/python3.7/configparser.py", line 1160, in _convert_to_boolean
raise ValueError('Not a boolean: %s' % value)
ValueError: Not a boolean: ('False',)
pi@raspberrypi: $
Envoyé

arthur5640, 2021-01-06:
Pibooth. Log is empty

DJ-Dingo, 2021-01-06:
Hey @arthur5640 Not shure if this is the same problem i had, but i downgraded pygame 2.0.0 to version 1.9.6
But can see you are using pygame 2.0.1, Have not tried that one

But this worked for me:
pip3 install pygame==1.9.6

werdeil, 2021-01-06:
Hi I didn't really understood 

> " when i put on escape twice"

But from the traceback it seems that your issue comes from the autostart feature, can you open the config file stored in `~/.config/pibooth/pibooth.cfg` and check the value of the key `[GENERAL][autostart]`, it shall be either `True` or `False` and it seems that it is something like ('False',)

arthur5640, 2021-01-07:
I don't find ~/.config/pibooth/pibooth.cfg

werdeil, 2021-01-07:
Can you open a shell and run
> pibooth --config
It shall open the file with a text editor then check the value of `[GENERAL] [autostart]`

Or if you want you can start over your config with the following command:
> pibooth --reset

Tell me if it resolve your issue.

arthur5640, 2021-01-07:
[image]

webmin81, 2021-01-09:
same problem ... please help thank you!

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pygame==1.9.6
  Using cached https://www.piwheels.org/simple/pygame/pygame-1.9.6-cp37-cp37m-linux_armv7l.whl
Installing collected packages: pygame
  Found existing installation: pygame 2.0.1
    Uninstalling pygame-2.0.1:
      Successfully uninstalled pygame-2.0.1
Successfully installed pygame-1.9.6
pi@fotobox:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 425, in main
    config = PiConfigParser("~/.config/pibooth/pibooth.cfg", plugin_manager)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 256, in __init__
    self.load()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 294, in load
    self.handle_autostart()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 308, in handle_autostart
    enable = self.getboolean('GENERAL', 'autostart')
  File "/usr/lib/python3.7/configparser.py", line 828, in getboolean
    raw=raw, vars=vars, fallback=fallback, **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 808, in _get_conv
    **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 802, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib/python3.7/configparser.py", line 1160, in _convert_to_boolean
    raise ValueError('Not a boolean: %s' % value)
ValueError: Not a boolean: ('False',)
pi@fotobox:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 425, in main
    config = PiConfigParser("~/.config/pibooth/pibooth.cfg", plugin_manager)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 256, in __init__
    self.load()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 294, in load
    self.handle_autostart()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/config/parser.py", line 308, in handle_autostart
    enable = self.getboolean('GENERAL', 'autostart')
  File "/usr/lib/python3.7/configparser.py", line 828, in getboolean
    raw=raw, vars=vars, fallback=fallback, **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 808, in _get_conv
    **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 802, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib/python3.7/configparser.py", line 1160, in _convert_to_boolean
    raise ValueError('Not a boolean: %s' % value)
ValueError: Not a boolean: ('False',)

webmin81, 2021-01-09:
ok sudo pibooth was the solution ... then it works :) thanks for the great photo box

sceric1235, 2021-01-16:
Hi,
i have the same problem. I debugged a little bit and found out, that in menu.py in function _on_selector_changed the values are wrong, so after saving the config file it will be "destroyed" by wrong values.
E.g.:
When i switch the language to DE the value in config file is ('DE',) after saving. The same with all the other values.
Do you know, where this failure is coming from?
Best wishes Eric

sceric1235, 2021-01-17:
Hi again. I found out, whats the problem. Its a change in pygame-menu bib in version 3.4.0 (see changelog: [https://www.pygame.org/project/3165/7934](https://www.pygame.org/project/3165/7934)
"... Fixed Selector value in get_value() method. Now the entire tuple is returned instead of [0] item ff217d7. ..."
After downgrading to version 3.3.2 `sudo pip3 install pygame-menu==3.3.2` it works.

werdeil, 2021-01-17:
Indeed you are right this comes from a change in `pygame-menu` @anxuae shall we force the `pygame` and `pygame-menu` version in install to avoid this kind of issues?

charly2203, 2021-01-19:
it still doesn't work on my side.
I downgraded pygame to 1.9.6 and pygame-menu to 3.3.2. Is that correct ?

sceric1235, 2021-01-19:
Hi @charly2203 
thats correct. After crashing the config file you have to correct it or remove it, before starting pibooth again. It should work then and pygame-menu should not destroy the file again. You can find the file at ~/.config/pibooth/pibooth.cfg

charly2203, 2021-01-19:
It works !
You're awesome !

If you update the project. I also had to install this package to make it works after a fresh install:
sudo apt-get install python3-sdl2

Maybe you could add this too, to the project ? (Sorry I'm not able to do that)

werdeil, 2021-01-20:
> It works !
> You're awesome !
> 
> If you update the project. I also had to install this package to make it works after a fresh install:
> sudo apt-get install python3-sdl2
> 
> Maybe you could add this too, to the project ? (Sorry I'm not able to do that)

The package `python3-sdl2` is only necessary when using `pygame` v2 on a raspberry pi. We cannot add it for everyone.

closed 2021-01-27 by arthur5640

anxuae, 2021-01-31:
> Indeed you are right this comes from a change in pygame-menu @anxuae shall we force the pygame and pygame-menu version in install to avoid this kind of issues?

 @werdeil I agree, It seems that regularly compat is broken between different version of `pygame-menu`, I will fix it