pibooth/pibooth #213: 3 problems with pibooth

rocsila, 2021-01-19:
Goodnight.  I do not know if it is appropriate to publish it here, but I would be very grateful if you could help me solve these 3 problems:

1. The program starts but when taking the photo it is blocked and the errors of the attached photo appear.
[image]
 
  -Invalid value ‘100’ for option iso. 
  - Unsupported option imaettings/ iso 100. 

2. I don't know how to change the language of the text to Spanish.  It is also my first project in raspberry.

3. How to get the program to start only when you turn on the raspberry.

werdeil, 2021-01-20:
Hi @rocsila,

1. Can you tell us which camera you are using? It seems that it is not fully supported by gphoto2.
2. and 3. You can access to the settings menu by pressing "esc" button on the main screen of pibooth, you can find the language option and the startup option too.

rocsila, 2021-01-20:
Good morning:
 Thank you very Mach for answering me so soon @werdeil. 
My camera is a powershot sx540 that I have tested with gphoto2 and I can shoot and access its folders.  
The problem with pibooth is the error that I get: it allows to shoot the photo but the program is blocked and does not get to process the photo, or print it.  it just stays fixed and locked.

Invalid value ‘100’ for option iso.
Unsupported option imaettings/ iso 100.
Opps, It seems that pibooth has crached.

anxuae, 2021-04-24:
Hi @rocsila 

Do you still have some issue with your camera?

- If yes, please attach the file generated by the command: `pibooth-diag`
- If no, close the issue

BR
anxuae

anxuae, 2021-05-11:
No news since long time so I close it.
Please re-open it if you are still stuck on this issue.

closed 2021-05-11 by anxuae