pibooth/pibooth #215: Pibooth hangs when taking the photo

rocsila, 2021-01-22:
Good afternoon: 
I'm new to this and I'm slowly making progress with pibooth, but it is impossible for me to solve this problem.  
I have a Powershot sx540 that works with gphoto2 but when I take the photo in pibooth the program hangs and I get these errors:

pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ WARNING ] pibooth           : No printer configured in CUPS (see http://localhost:631)
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ ERROR   ] pibooth           : Unsupported option actions/viewfinder=0 ([-7] I/O problem), configure your DSLR manually
[ ERROR   ] pibooth           : [-7] I/O problem
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.4.83-v7+', version='#1379 SMP Mon Dec 14 13:08:57 GMT 2020', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

I don't know how to solve them (I have changed settings on the camera and it is impossible).  I would really appreciate your help becarse I am losing my mind. 

Thanks!!!

werdeil, 2021-01-27:
Hi @rocsila 

Is the Powershot sx540 mirror
The issue "Unsupported option actions/viewfinder=0" is commonly linked to it, but @anxuae made a change on this, I'm not sure it is included on the last version.

anxuae, 2021-05-06:
Hi @rocsila, some news about this topic?
If the your problem is solved, can you close the issue?

s5919, 2022-10-30:
Hi @anxuae,
I have exactly the same issue in my setup. I am also using the Canon PowerShot SX540 as well. Taking the picture is possible but after that the pibooth application is freezing and after around 60 seconds the same error message is shown. Can you please help me to solve the issue?

Here is my log file:
[pibooth.log](https://github.com/pibooth/pibooth/files/9895990/pibooth.log)

werdeil, 2022-10-30:
Yes it is the same issue, which comes from the action send by photo to your DSLR: 
```
Unsupported option actions/viewfinder=0 ([-7] I/O problem), configure your DSLR manually
```
The `viewfinder` action is there to move the mirror, can you confirm your DSLR is mirrorless?

s5919, 2022-10-30:
Yes I can confirm that it is mirrorless. Is it possible to switch to a mirrorless camera via some settings or do I need to step through the code and uncomment lines that try to move the mirror?

anxuae, 2022-10-31:
Even if it is mirror less, the viewfinder is not the issue (viewfinder option will not be set, that all)

You have an IO issue between DSLR and pibooth, which generally means that DSLR is busy
Can you confirm that you have disabled autofocus?

s5919, 2022-10-31:
Yes I have deactivated the autofocus and tried to deactivate all other kind of settings that could cause this issue. Still the same problem. After taking the picture the camera seems busy.

If I just run "gphoto2 --capture-image" I can take multiple images without any issue.

s5919, 2022-11-20:
If I comment out the command "self.set_config_value('actions', 'viewfinder', 0)" in the capture function in gphoto.py, I am getting the error "[-110] I/O in progress". It looks like the camera is still busy with the preview stuff and cannot take the picture as requested in the line "self._captures.append((self._cam.capture(gp.GP_CAPTURE_IMAGE), effect))".

Any idea how to fix that?

flammeur, 2023-02-14:
Hello,

Just in case i got the same issue .

The SD card on the camera was full , got resolved after cleaning it

anxuae, 2023-05-06:
Hi @flammeur,

Thanks for reporting. I'm agree that this issue is coming from the DSLR. Not directly pibooth.
I'm closing this old issue for now.

closed 2023-05-06 by anxuae