pibooth/pibooth #216: Questions

arthur5640, 2021-01-27:
Hello, is it possible to change the layout icons or the phrase: take a photo?

DJ-Dingo, 2021-01-27:
@arthur5640 

In the command prompt: use

pibooth --fonts (get list of fonts)
pibooth --config (or use esc in pibooth to get into menu)
pibooth --translate (change the phrases, of every language, or make you own)

arthur5640, 2021-01-27:
Thanks but i want change this : 
[image]

werdeil, 2021-01-27:
The image you want to change is only used on the README of the repository. if you want it to change you'll need your own repository or a fork and then you'll have the right to do it on this new repo.

But I'm sure this is what you are asking, for the rest, as said by @DJ-Dingo you have some options to change the texts, for the layouts/images these are more difficult to change.

DJ-Dingo, 2021-01-27:
Ok. 

The text you can change with pibooth --translate 
The pictures you can change under the folder assets, but if you re-install it will proberly overwrite them again. 
 /home/pi/.local/lib/python3.7/site-packages/pibooth/pictures/assets

arthur5640, 2021-01-27:
Ok thansk. I try to copy. png in pictures/assets but acced is denied.

DJ-Dingo, 2021-01-27:
> Ok thansk. I try to copy. png in pictures/assets but acced is denied.

What you want is not a normal thing to do in pibooth, and the best is to do like @werdeil said "make your own repository or a fork" to your own github. 

I have tryed what you want myself, just for fun. By overwritten the files with my own set of files with the same names.
but it´s not the correct way of doing it.

Maybe you need some sudo. Are you using command prompt or Raspberry Pi Desktop, then you can give yourself access or chmod files [here som chmod info](https://www.dummies.com/computers/raspberry-pi/working-with-file-permissions-on-your-raspberry-pi/)

If you are new to this, i suggest you see some youtube video on linux - file-permissions.

arthur5640, 2021-01-27:
I dont know, what is a fork.
Thanks

DJ-Dingo, 2021-01-27:
@arthur5640  I will not be able to explain it all to you here. You will need to google it or read the github docs, maybe watch some youtube. Found a few links and a video you can watch. But it sounds like it's a bit too much for you to get started with. 

https://www.youtube.com/watch?v=f5grYMXbAV0
https://docs.github.com/en/github/getting-started-with-github/fork-a-repo
https://deanmalone.net/post/how-to-fork-your-own-repo-on-github/

arthur5640, 2021-01-27:
Thanks. I read this

anxuae, 2021-04-24:
I think that this question can be closed now.
Please open a new one if you have more technical questions on pibooth

closed 2021-04-24 by anxuae