pibooth/pibooth #217: pibooth --reset - not working with "sound effects" plugin installed

DJ-Dingo, 2021-01-27:
There is a problem with "pibooth --reset" when you have installed the "pibooth_sound_effects" plugin.

pi@raspberrypi:~ $ pibooth --reset
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Generate the translation file in '/home/pi/.config/pibooth/translations.cfg'
[ INFO    ] pibooth           : Installed plugins: lcd_i2c_extra-1.0.0, sound-effects-0.0.3, extra-lights-1.0.1, picture-template-0.0.3
[ INFO    ] pibooth           : Generate the configuration file in '/home/pi/.config/pibooth/pibooth.cfg'
Traceback (most recent call last):
  File "/home/pi/.local/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/booth.py", line 453, in main
    plugin_manager.hook.pibooth_reset(cfg=config, hard=True)
  File "/home/pi/.local/lib/python3.7/site-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/home/pi/.local/lib/python3.7/site-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/home/pi/.local/lib/python3.7/site-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/home/pi/.local/lib/python3.7/site-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth_sound_effects/pibooth_sound_effects.py", line 28, in pibooth_reset
    shutil.copytree(source_sound_path, sound_path)
  File "/usr/lib/python3.7/shutil.py", line 321, in copytree
    os.makedirs(dst)
  File "/usr/lib/python3.7/os.py", line 221, in makedirs
    mkdir(name, mode)
FileExistsError: [Errno 17] File exists: '/home/pi/.config/pibooth/sounds'

DJ-Dingo, 2021-01-27:
If i delete the "sounds" folder first, it works, 
but now it missing some languages.

werdeil, 2021-01-27:
I would say it is an issue on the plugin side, I'll try to have a look when I get time.

DJ-Dingo, 2021-01-27:
Yeah, you right, and about the languages part, i re downloadet the languages.py and now it seems to work.

An idea, when you use pibooth --reset 
It could be nice to have it make a backup of translations.cfg and config.cfg like "config.cfg_datetime" in a folder. That way you can go in and see previous setups,  like GPIO setup, Template link, Plugin Link - for faster re.setup

anxuae, 2021-04-24:
Hi,

I'm moving this issue to the plugin `pibooth-sound-effects`.

I will create a new issue with your idea of backup files.