pibooth/pibooth #218: pibooth with mxLinux

fetze82, 2021-02-01:
Hello,
is it possible to use this project with mxLinux.

I installed all in the manual. But i get problems with

$ $ sudo pip3 install pibooth[dslr,printer]
bash: $: Kommando nicht gefunden.

i tryed this to
Chris@Chris:~/Desktop/pibooth-2.0.1
$ python setup.py
Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from setuptools import setup, find_packages
ImportError: No module named setuptools

Please could somebody help?

Thanks
Greatings
Chris

fetze82, 2021-02-01:
Hey,
i get the nex step - could someone help to get it running

Chris@Chris:~/Downloads/PiBooth2/pibooth
$ pip3 install pibooth
Collecting pibooth
  Using cached https://files.pythonhosted.org/packages/12/11/0d626aa391bc158c402aba905f479c17b1e18f4805ec210d313fd90bfc93/pibooth-2.0.1-py2.py3-none-any.whl
Collecting pygame-menu>=3.1.0 (from pibooth)
  Downloading https://files.pythonhosted.org/packages/d1/91/cd695896580490ff788fce840615a4b74da5dc585165e2429d62de146fad/pygame_menu-3.5.5-py2.py3-none-any.whl (804kB)
    100% |████████████████████████████████| 808kB 805kB/s 
Collecting pluggy>=0.13.1 (from pibooth)
  Downloading https://files.pythonhosted.org/packages/a0/28/85c7aa31b80d150b772fbe4a229487bc6644da9ccb7e427dd8cc60cb8a62/pluggy-0.13.1-py2.py3-none-any.whl
Collecting Pillow>=6.1.0 (from pibooth)
  Downloading https://files.pythonhosted.org/packages/eb/8e/d2f7a67cf8da9b83c1e3ee38dbf49448f3c8acb2cb38f76e4301f4a70223/Pillow-8.1.0-cp37-cp37m-manylinux1_x86_64.whl (2.2MB)
    100% |████████████████████████████████| 2.2MB 307kB/s 
Collecting pygame>=1.9.6 (from pibooth)
  Downloading https://files.pythonhosted.org/packages/4b/9e/c400554dd1d0e562bd4379f35ad5023c68fc120003a58991405850f56f95/pygame-2.0.1-cp37-cp37m-manylinux1_x86_64.whl (11.8MB)
    100% |████████████████████████████████| 11.8MB 50kB/s 
Collecting gpiozero>=1.5.1 (from pibooth)
  Downloading https://files.pythonhosted.org/packages/5a/06/e3440c452a0c2ba1f12527c12c5676b158f2591ac6c43979016f21f71b2a/gpiozero-1.5.1-py2.py3-none-any.whl (138kB)
    100% |████████████████████████████████| 143kB 3.4MB/s 
Collecting psutil>=5.5.1 (from pibooth)
  Using cached https://files.pythonhosted.org/packages/e1/b0/7276de53321c12981717490516b7e612364f2cb372ee8901bd4a66a000d7/psutil-5.8.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-j30_uddh/psutil/

Thanks
Greatings
Chris

werdeil, 2021-02-05:
Hi @fetze82, you are missing the python3-setuptools package. I don't know how you can install it on MXLinux, on a Debian it would be simply:
> sudo apt-get install python3-setuptools

fetze82, 2021-02-05:
Hey,
thanks for the information

i did a few steps (maybe one more, or one less)
sudo apt-get install python3-setuptools
sudo apt-get install build-essential python-dev python-pip
pip3 install -U wheel --user
sudo apt install python3-psutil

after that i was able to install pibooth
pip3 install pibooth

i could start it. Now i tried to do change the language and quite the applikation. I got an error

Chris@Chris:~
$ /home/Chris/Desktop/pibooth-2.0.1/pibooth/pibooth
pygame 2.0.1 (SDL 2.0.14, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application without physical GPIO, fallback to GPIO mock
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ INFO    ] pibooth           : Generate the configuration file in '/home/Chris/.config/pibooth/pibooth.cfg'
[ ERROR   ] pibooth           : Not a boolean: ('False',)
Traceback (most recent call last):
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/booth.py", line 365, in main_loop
    self._menu.process(events)
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/config/menu.py", line 245, in process
    self._main_menu.update(events)
  File "/home/Chris/.local/lib/python3.7/site-packages/pygame_menu/menu.py", line 1962, in update
    elif self._current._menubar.update(events):
  File "/home/Chris/.local/lib/python3.7/site-packages/pygame_menu/widgets/widget/menubar.py", line 372, in update
    self.apply()
  File "/home/Chris/.local/lib/python3.7/site-packages/pygame_menu/widgets/core/widget.py", line 387, in apply
    return self._on_return(*args, **self._kwargs)
  File "/home/Chris/.local/lib/python3.7/site-packages/pygame_menu/menu.py", line 1290, in _back
    self._close()
  File "/home/Chris/.local/lib/python3.7/site-packages/pygame_menu/menu.py", line 1561, in _close
    onclose()
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/config/menu.py", line 199, in _on_close
    self.cfg.save()
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/config/parser.py", line 288, in save
    self.handle_autostart()
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/config/parser.py", line 308, in handle_autostart
    enable = self.getboolean('GENERAL', 'autostart')
  File "/usr/lib/python3.7/configparser.py", line 828, in getboolean
    raw=raw, vars=vars, fallback=fallback, **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 808, in _get_conv
    **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 802, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib/python3.7/configparser.py", line 1160, in _convert_to_boolean
    raise ValueError('Not a boolean: %s' % value)
ValueError: Not a boolean: ('False',)
[ ERROR   ] pibooth           : system='Linux', node='Chris', release='4.19.0-13-amd64', version='#1 SMP Debian 4.19.160-2 (2020-11-28)', machine='x86_64', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

after that i wasn`t able to start it again
$ /home/Chris/Desktop/pibooth-2.0.1/pibooth/pibooth
pygame 2.0.1 (SDL 2.0.14, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/home/Chris/Desktop/pibooth-2.0.1/pibooth/pibooth", line 10, in <module>
    sys.exit(main())
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/booth.py", line 425, in main
    config = PiConfigParser("~/.config/pibooth/pibooth.cfg", plugin_manager)
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/config/parser.py", line 256, in __init__
    self.load()
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/config/parser.py", line 294, in load
    self.handle_autostart()
  File "/home/Chris/.local/lib/python3.7/site-packages/pibooth/config/parser.py", line 308, in handle_autostart
    enable = self.getboolean('GENERAL', 'autostart')
  File "/usr/lib/python3.7/configparser.py", line 828, in getboolean
    raw=raw, vars=vars, fallback=fallback, **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 808, in _get_conv
    **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 802, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib/python3.7/configparser.py", line 1160, in _convert_to_boolean
    raise ValueError('Not a boolean: %s' % value)
ValueError: Not a boolean: ('False',)

Maybe someone could help me.
sorry for the noop questions

Greatings Chris

anxuae, 2021-04-24:
Hi @fetze82 

Did you solve your issue? I guess it is due to the pygame-menu package version.
Please try py executing command: `pip3 install pygame-menu==3.5.5`

anxuae, 2021-05-06:
I'm closing this issue because it is related to the pygame-menu version.
In the next ``pibooth`` version, the ``pygame-menu`` version will be fixed to avoid wrong version installation

closed 2021-05-06 by anxuae