pibooth/pibooth #219: Touchscreen and pygame menu not working properly

qchatenet, 2021-02-05:
Hi,

I noticed two problems recently : 
- touchscreen is not working with pygame version 2.0.1 (it is working with 1.9.6 version)
- when I exit the menu after doing no modification, pibooth crashes because of several fields in pibooth.cfg  have been changed (eg. : language = ('fr',) instead of language = fr ) Happened with both 2.0.1 and 1.9.6 pygame versions.

Thanks for the help.

werdeil, 2021-02-05:
Indeed:
https://github.com/pibooth/pibooth/issues/201
In pygame 2 the touchscreen is no longer considered as a mouse hence some change to do (some have already been done but not merged)
https://github.com/pibooth/pibooth/pull/202

https://github.com/pibooth/pibooth/issues/212
The crash comes from a change of behavior in pygame-menu, it was fixed by @anxuae 5 days ago, but not yet released.

qchatenet, 2021-02-05:
I am looking forward to the next release. Thank you for the update.

gliepins, 2021-04-26:
What is the solution to get the touchscreen working? If the py-game needs to be of earlier version, how to achieve that? A regular install command installs the latest. 
Thanks for help!

anxuae, 2021-04-27:
The touchscreen is not working for the moment.
I'm working on finalizing this feature. Probably included in the next release (next week?)

detrist, 2021-05-05:
Previously no but now I noticed that it work, at least outside of the menu.

anxuae, 2021-05-07:
Fixed in `pibooth` v2.0.2

BR
anxuae

closed 2021-05-07 by anxuae