pibooth/pibooth #220: Rotate has no effect.

dafunkychild, 2021-02-18:
Hello, I did follow the instructions described in issue #196 but modifying rotation parameter ( 90, 180, 270 ) in the config file as no effect.

**Environment**
 - OS Raspbian Buster
 - Camera DSLR Canon1200d or  Rebel T5 / Kiss X7 in other parts of the globe.
 - gphoto2 2.5.23.1
 - Pibooth version v2.0.1

werdeil, 2021-04-23:
Hi @dafunkychild, I can't try on my side as I don't have access to a setup with a DSLR but is the DSLR setter to not rotate the image on its side?

anxuae, 2021-04-25:
Hi @dafunkychild,

Indeed, this feature has never been implemented, except for RaspberryPi camera.
I implemented it in commit: e20a9e0a7d6f860df137baccf624410b0c79872f

It will be available in `pibooth` version 2.0.2 (release not yet planned)

closed 2021-04-25 by anxuae