pibooth/pibooth #221: Canon EOS 1100D [ DSLR camera]

GiorgioIOT, 2021-02-23:
sorry my english is not perfect.

problem with Canon Eos camera, Pibooth V.2.0.1 installed, crashes, I attach files with description.

`Pibooth version installed: 2.0.1
Listing all connected DSLR camera
00 : addr-> usb:001,005  name-> Canon EOS 1100D

================================================================================
Stating diagnostic of connected DSLR camera
* Preview compatible: True
* Capture compatible: True
/actions/syncdatetime
  Label       : Synchronize camera date and time with PC
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]
/actions/uilock
  Label       : UI Lock
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 2
  Choices     : [0, 1]
/actions/popupflash
  Label       : Popup Flash
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 2
  Choices     : [0, 1]
/actions/autofocusdrive
  Label       : Drive Canon DSLR Autofocus
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]
/actions/manualfocusdrive
  Label       : Drive Canon DSLR Manual focus
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Radio button
  Current     : None
  Choices     : ['Near 1', 'Near 2', 'Near 3', 'None', 'Far 1', 'Far 2', 'Far 3']
/actions/cancelautofocus
  Label       : Cancel Canon DSLR Autofocus
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]
/actions/eoszoom
  Label       : Canon EOS Zoom
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Text
  Current     : 0
  Choices     : n/a
/actions/eoszoomposition
  Label       : Canon EOS Zoom Position
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Text
  Current     : 0,0
  Choices     : n/a
/actions/viewfinder
  Label       : Canon EOS Viewfinder
[... 387 lines cut ...]
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
  Current     : 100
  Choices     : ['100', '0', '75', '0', '50']

================================================================================
Testing commands used by pibooth
Setting option imgsettings/iso="100"
   -> invalid value '100' for option iso (possible choices: ['Auto'])
   -> unsupported setting imgsettings/iso=100 (nothing configured on DSLR)
Setting option settings/capturetarget="Memory card"
Getting option actions/viewfinder=0
Setting option actions/viewfinder="1"
Take capture preview
Setting option actions/viewfinder="0"
Take a capture

================================================================================
ABORT   : exception occures: [-110] I/O in progress
If you are investigating why pibooth does not work with your DSLR camera,
please paste the content of generated file 'diagnostic.log'
on https://github.com/pibooth/pibooth/issues
`

the system is installed on Raspberry pi 4

werdeil, 2021-02-26:
Hi @GiorgioIOT I don't think the crash comes from the use of a Canon 1100D as we are using this model on our version. Can you tell us more about the crash? When does it happen?

GiorgioIOT, 2021-02-27:
> Ciao @GiorgioIOT Non credo che l'incidente derivi dall'uso di una Canon 1100D dato che stiamo usando questo modello sulla nostra versione. Puoi dirci di più sull'incidente? Quando succederà?

Hi Werdeil, I am using the Canon (Camera) in automatic mode.
Pibooth app in FullScreen mode, when I start I use button 1 to start the procedure, I choose between 4 photos or 1 photo, it starts with the vision of the Raspberry camera, I wait for the pause, take the photo with the Canon, I realize that it also starts the flas on the Canon, I await the vision for the press and goes to Oops.
I check on the Canon and on the sd card I find the picture taken.
Pygame version 1.9.6

werdeil, 2021-03-02:
Ok so the issue is coming when pibooth tries to download the file from the Canon, does it have a SD card in it, with enough space?

GiorgioIOT, 2021-03-03:
verifying SD from CANON and Raspberry, they have enough space to save photos.
I worked on the Canon setting only normal photos without RAW, I'm checking that it works. Do you recommend that you leave the Canon like this and lose the RAW photos?

werdeil, 2021-03-03:
Yes if the issue was that you used RAW, put only jpg ones, we don't use raws and it can indeed create issue when downloading them as we don't expect to have anything else than jpg.

GiorgioIOT, 2021-03-03:
I don't know what happened, but it doesn't work anymore, not even from Camera Raspberry !!!!!!
It worked for a while with the Canon, I installed Google Photo.
BUT I don't think that's the problem.
He saves the photo and then falls ....

werdeil, 2021-03-05:
We may need more info to help you with this new fail: can you provide us some logs?

GiorgioIOT, 2021-03-06:
I restored the pibooth.cfg file, I noticed that there are brackets and commas on the string and number entries, restoring it seems to work, I did not understand how to insert the google token on chat_id, is there any guide?

I realized that when I enter the pibooth graphic menu and exit this file is generated with brackets and commas.

werdeil, 2021-03-08:
Seems that you have the issue with the last pygame-menu version, we made a fix https://github.com/pibooth/pibooth/commit/0096604b09d65f7cae3880d78a0bb14defff27d0 but didn't release a new version yet.

You can either try to install the development version or downgrade only the pygame-menu module as discussed here https://github.com/pibooth/pibooth/issues/212#issuecomment-763044955

anxuae, 2021-04-24:
Hi @GiorgioIOT 

Did you solve your issue? If yes, please close this issue.

BR
anxuae

GiorgioIOT, 2021-04-28:
the problem was solved by installing Pygame as described by (werdeil)

closed 2021-04-28 by GiorgioIOT