pibooth/pibooth #222: Multiple images captured per one request

detrist, 2021-03-04:
Hello, I have just installed the latest pibooth on Raspberry PI 4B. I am just wondering if this is a normal behaviour that using a DSLR camera it always captures 4 photos per one image request.

I currently have some trouble with building and debugging the app because of a few dependency issues but will try to resolve them.

Kindly find the config and the logs here:
[pibooth.cfg.txt](https://github.com/pibooth/pibooth/files/6084781/pibooth.cfg.txt)
[pibooth.log](https://github.com/pibooth/pibooth/files/6084785/pibooth.log)

**Environment**
 - OS: Raspbian Buster
 - Camera: Nikon D3400
 - Pibooth version: 2.0.1

DJ-Dingo, 2021-03-04:
Your config file says captures = (4, 1)

Press ESC to open menu, go to "Pictures" - change "Number of captures" to 1 
Or chance the settings directly in your pibooth.cfg   captures = (4, 1)   to   captures = (1)

detrist, 2021-03-05:
Thank you for your feedback but my problem is not related to the captures feature. By forcing the number of captures to 1 it still shoots the camera 4 times and the final image contains the last one. Just to explain what I noticed, if I choose the number of captures to 4, then it shoots the camera 16 times and the final image contains all the 4th images.

After some hours with debugging I realized that the problem I have is with the preview feature and probably with the openCV implementation. The behaviour disappeared after I removed the related calls from the camera plugin. May I request some explanation about how the preview function expect to work?

werdeil, 2021-03-05:
Indeed if you are using only a DSLR the preview is in fact taking photos as fast as possible to simulate a video, hence the multiple shot sounds.
If I remember well if  gphoto is able to control the mirror of your DSLR it will keep it up for all the previews (and you won't hear the mirror moving sound).

detrist, 2021-03-05:
It makes sense, thank you for the explanation. Then I will try to force the app to use gphoto for preview. Ticket can be closed.

closed 2021-03-05 by detrist