pibooth/pibooth #223: PiBooth for portrait screen resolutions

balushen284, 2021-03-14:
Hello
I am currently preparing my wedding and would like to make a photobooth based on PiBooth similar to this Magic Mirror: [https://www.pinterest.com/pin/322640760797627787/](url)

My version would be much more simple and the options given by PiBooth are enought for me.

I have been playing with the options on PiBooth config, and although I can change the resolution, if I set a "portrait" resolution as 1024x1280 the asset images (take 1 photo, take 4 photos, final view and so on), get overlapped or cropped as I guess that PiBooth is not ready for portrait mode and is only currently intended for a landscape screen resolution.

Is there any way I could set portrait mode for PiBooth? If not, could you provide me some with some hints about where I should look at on the code to try to make a portrait mode PiBooth fork? I am not an expert on programming but could try to do my best.

Thanks in advance for your help

anxuae, 2021-03-14:
Hi @balushen284 ,

The default configuration of pibooth is working in portrait.
Pibooth works in landscape or portrait mode, you can find the way to change orientation here: https://github.com/pibooth/pibooth#final-picture-rendering

balushen284, 2021-03-15:
Hi @anxuae 
Thanks for your response, but I do not mean the final picture.
The problem I see is with the content on the window PiBooth opens if I set a portrait resolution.
I have been playing with background.py and window.py and changing sizes on class ChooseBackground, IntroBackgroundWithPicture and so on and I managed to fit everything.

For example if on background.py on class ChooseBackground, resize function I change this line:

size = (self._rect.width * 0.6, self._rect.height * 0.6)

To

size = (self._rect.width * 0.4, self._rect.height * 0.4)

And set a window size for PiBooth of (400,800) (portrait size for a window), the images related to layout selection (1 photo, 2 photos,...), now do not overlap.

Maybe the way to solve this would be to include a new attribute on PiBooth config and change this multipliers depending on if It is a portrait or landscape window size for PiBooth.

Regards

werdeil, 2021-03-15:
Indeed we never tested pibooth with a portrait screen. If you have found a way to tweak it feel free to push a pull request we could try to integrate it.

I may try to work on it but not now (I'm quite off  coding currently).

anxuae, 2021-04-28:
Hi,

I've worked on this issue and I fixed the case of the `ChooseBackground` background in #231.
Can you tell me what are the other background with an overlapping? (to be sure that I don't missed any)

anxuae, 2021-05-07:
Fixed in `pibooth` v2.0.2

closed 2021-05-07 by anxuae

Aurel-Charles, 2023-04-11:
Hi guys! I'm having the same issue, I'm using a 1920*1080 touchscreen and want to use it in portrait mode.
I put it in fullscreen mode but it still behave as landscape and crop everything on the right.
I try to put the window size in the config to 1080*1920 . Its "ok" but I'm still having the menu bar to deal with. And it also behave like in landscape during the capture time.
Im using the 2.0.7 version
Any Idea on how to fix it? 
Thanks