pibooth/pibooth #224: No Camera detected over wifi

iak92, 2021-03-29:
**Describe the bug**
I want to use pibooth with my lumix camera. It uses a wifi interface and http comands. It is integrated in libgphoto2 since 5.27 and I can use the camera wit gphoto2 and gtkam. When I start pibooth it says no camera detected.

**Steps to reproduce the bug**
1. Start Pibooth over console
2. get no camera detected

**Logs**
`
Traceback (most recent call last):
  File "/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/pibooth/booth.py", line 456, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/lib/python3.9/site-packages/pibooth/booth.py", line 108, in __init__
    self.camera = camera.get_camera(config.getint('CAMERA', 'iso'),
  File "/usr/lib/python3.9/site-packages/pibooth/camera/__init__.py", line 37, in get_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected
`

**Environment**
 - Laptop with Arch Linux
 - Camera: Panasonic Lumix G70 
 - Pibooth version 
 `pygame 2.0.1 (SDL 2.0.14, Python 3.9.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.1
2.0.1
`

werdeil, 2021-04-23:
Hi @iak92, I switch the label to enhancement as we never tried to used a photo camera over Wifi so this cannot be considered as a bug.

Unfortunately we don't have much time to work on adding this new feature right now but I'll try to see if I could find a quick fix.

anxuae, 2021-04-24:
Hi @iak92,
Can you past the file generated by renting the command: `pibooth-diag`

iak92, 2021-05-15:
Hi,
this is the output of `pibooth-diag`
`
pygame 2.0.1 (SDL 2.0.14, Python 3.9.5)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.4
Pibooth version installed: 2.0.2
Installed plugins: 
Listing all connected DSLR camera
No compatible DSLR camera detected`

anxuae, 2021-05-15:
Hi @iak92 

Can you give me the result of the gphoto2 command:

```
gphoto2 --auto-detect
```

If no camera  detected, I don't understand how gtkam is able to detect it. (maybe gtkam use an other driver lib than gphoto2?)

iak92, 2021-05-15:
auto-detect returns an empty list.
I connected manually, so I choose camare type in gtkam and configured the ip-adress. For the gphoto2 cli I also have to define the protocol to ip and give the ip-adress

anxuae, 2021-05-15:
Can you give me the command of the gphoto2 cli you are using? (sorry but I have no IP camera, so I can not test)

I guess something like:

```bash
gphoto2 --capture-image-and-download --port ptpip:10.5.5.9
```

anxuae, 2021-05-22:
Hi @iak92,

using the new hook pibooth_setup_camera to customize the camera in a plugin, you should be able to manage custom configuration for your camera (see example on https://github.com/pibooth/pibooth/blob/master/docs/plugin.rst#example-5--setup-a-custom-camera)

anxuae, 2021-07-15:
Please reopen the issue if you have more information to provide about the camera configuration and settings

closed 2021-07-15 by anxuae