pibooth/pibooth #225: Problem with QR Code plugin

gliepins, 2021-04-09:
Guys, Hi everyone!
I cannot get around the qrcode plugin. 
I upgraded the pibooth to latest,
I installed the plugin sucessfuly,
now running the pibooth app it throws an error!
Please, help! @sravel @anxuae 

Here is the error below:

#pygame-menu 4.0.2
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 422, in main
    plugin_manager = create_plugin_manager()
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/__init__.py", line 19, in create_plugin_manager
    plugin_manager.load_setuptools_entrypoints(hookspecs.hookspec.project_name)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 299, in load_setuptools_entrypoints
    plugin = ep.load()
  File "/usr/local/lib/python3.7/dist-packages/importlib_metadata/__init__.py", line 167, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pibooth_qrcode'

```

werdeil, 2021-04-23:
Hi @gliepins unfortunately your traceback doesn't give us much idea on the issue. Have you tried removing the plugin and check if pibooth is working again? It will help us to see if the issue is on the plugin or the core app.

gliepins, 2021-04-23:
Hi!
It is a fresh install on raspberry. The application (the core app) runs fine without the plugin. What info would be useful?

anxuae, 2021-04-24:
Hi @gliepins

Did you install pibooth-qrcode using the command `sudo pip3 install pibooth-qrcode` ?

gliepins, 2021-04-24:
Hi! 
Yes exactly like that! And the first run of pibooth gives such error!

anxuae, 2021-04-24:
Can you paste here the result of the commande:

`pip3 freeze`

gliepins, 2021-04-24:
pi@raspberrypi:~ $ pip3 freeze
asn1crypto==0.24.0
astroid==2.1.0
asttokens==1.1.13
automationhat==0.2.0
beautifulsoup4==4.7.1
blinker==1.4
blinkt==0.1.2
buttonshim==0.0.2
cachetools==4.2.1
Cap1xxx==0.1.3
certifi==2018.8.24
chardet==3.0.4
Click==7.0
colorama==0.3.7
colorzero==1.1
cookies==2.2.1
cryptography==2.6.1
cupshelpers==1.0
docutils==0.14
drumhat==0.1.0
entrypoints==0.3
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==1.0.2
fourletterphat==0.1.0
google-auth==1.28.1
google-auth-oauthlib==0.4.4
gpiozero==1.5.1
html5lib==1.0.1
idna==2.6
importlib-metadata==3.10.0
isort==4.3.4
itsdangerous==0.24
jedi==0.13.2
Jinja2==2.10
keyring==17.1.1
keyrings.alt==3.1.1
lazy-object-proxy==1.3.1
logilab-common==1.4.2
lxml==4.3.2
MarkupSafe==1.1.0
mccabe==0.6.1
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.3
mypy==0.670
mypy-extensions==0.4.1
numpy==1.16.2
oauthlib==2.1.0
olefile==0.46
pantilthat==0.0.7
parso==0.3.1
pexpect==4.6.0
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
pibooth==2.0.1
pibooth-google-photo==1.0.2
pibooth-qrcode==0.0.1
picamera==1.13
piglow==1.2.5
pigpio==1.78
Pillow==8.2.0
pluggy==0.13.1
psutil==5.5.1
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycairo==1.16.2
pycrypto==2.6.1
pycups==1.9.73
pygame==2.0.1
pygame-menu==4.0.2
Pygments==2.3.1
PyGObject==3.30.4
pyinotify==0.9.6
PyJWT==1.7.0
pylint==2.2.2
pyOpenSSL==19.0.0
pyperclip==1.8.2
pyserial==3.4
pysmbc==1.0.15.6
python-apt==1.8.4.3
pyxdg==0.25
qrcode==6.1
rainbowhat==0.1.0
reportlab==3.5.13
requests==2.21.0
requests-oauthlib==1.0.0
responses==0.9.0
roman==2.0.0
RPi.GPIO==0.7.0
rsa==4.7.2
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
Send2Trash==1.5.0
sense-hat==2.2.0
simplejson==3.16.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
soupsieve==1.8
spidev==3.4
ssh-import-id==5.7
thonny==3.3.6
touchphat==0.0.1
twython==3.7.0
typed-ast==1.3.1
typing-extensions==3.7.4.3
unicornhathd==0.0.4
urllib3==1.24.1
webencodings==0.5.1
Werkzeug==0.14.1
wrapt==1.10.11
zipp==3.4.1