pibooth/pibooth #226: Relay module use for light during capture sequence

jopar59, 2021-04-09:
Hello,

I think it could be a good idea to pilot a light ON/OFF with a simple relay module connected during the capture sequence.

Hope it could be done :)

Thank you for the app, it is great :)

DJ-Dingo, 2021-04-10:
You can allready do that. I use a relay for all my external lights and internal lights. (You can even have more than 1 light activatet on the same time by edit the pins in the config see example below  Etc.   flash_led_pin = 33, 35

Edit the configuration by running the command pibooth --config.

Just install the pibooth-extra-lights, connect a relay and real lights with 12v, or 110v, or 230v 
https://pypi.org/project/pibooth-extra-lights/

Configuration
Here below the new configuration options available in the pibooth configuration. The keys and their default values are automatically added to your configuration after first pibooth restart.

The physical pin numbering is used.

[CONTROLS]

# Physical GPIO OUT pin to light a LED at pibooth startup (list of pins accepted)
startup_led_pin = 29

# Physical GPIO OUT pin to light a LED during the entire capture sequence (list of pins accepted)
preview_led_pin = 31

# Physical GPIO OUT pin to light a LED when the capture is taken (list of pins accepted)
flash_led_pin = 33

jopar59, 2021-04-10:
Hello,
Ok thank you :) I will try during the day

werdeil, 2021-04-23:
Hi @jopar59 did you try the pibooth-extra-lights plugin? did it fit your need?

jopar59, 2021-04-23:
Hello, Yes Perfect

DJ-Dingo, 2021-04-23:
@jopar59  If it works, you should use "Close issue" the button before "Comment" :-)

jopar59, 2021-04-23:
Done, thank you :)

closed 2021-04-23 by jopar59