pibooth/pibooth #227: Printed document has shifts in colors

detrist, 2021-04-17:
Hi All,

Not sure whether it is a bug or not but I have a really weird situation what so far I have not resolved. I have a HiTi P525L printer which seems working fine from Windows 10 by printing images. 

For some reason on RPI4 the printed images from Pibooth have shifts in colors along the X axis. The same happens if I want to print the image from shell so this is not the problem of the Pibooth code for first look. However, the the result is fine if I ask GThumb on RPI4 to print them same picture without changing its default settings. Definitly GThumb does something more which makes it work.

Have you ever had similar issue about the printing? Any suggestions would be appriciated.

Thanks for your feedback in advance.

anxuae, 2021-04-24:
Hi @detrist

It seems an issue concerning the support of your printer in CUPS (print server not provided by pibooth).
Did you try to install/update [GutenPrint](http://gimp-print.sourceforge.net/index.php).

``
sudo apt-get install printer-driver-gutenprint
``

Did you tray to change the settings of your printer in the [CUPS admin page](http://localhost:631/printers/)?

BR
anxuae

detrist, 2021-05-05:
Hi @anxuae,

I will try it again with a fresh PI installation but for first look it did not help and so I decided to move the printing part into Windows temporarly by providing a Rest API and call it instead of CUPS. Meanwhile we are in contact with the printer support provider if they could provide us a better driver.

I will update this issue later I have any.

Many thanks, detrist

detrist, 2021-05-05:
Better late than never, I finally managed to make the printing work with the gutenprint driver, but there were some tricks in its installation steps:
1.) Probably optional but I deleted the "printer-driver-gutenprint" package I found in the RPI repo and then downloaded the latest snapshot version from SourceForge[https://sourceforge.net/projects/gimp-print/files/snapshots/](url)
2.) I had to delete the official CUPS drivers and filters from the disk I previously installed

Also I noticed that the official driver contained a PPD file which was made with CUPS 1.4 and during the build of the driver a lot of deprication notice appeared. Still waiting for update from the product support.

Thank you for your assistance.

closed 2021-05-06 by anxuae