pibooth/pibooth #228: Make backup files using --reset option

anxuae, 2021-04-24:
Idea from @DJ-Dingo.

when you use pibooth --reset
It could be nice to have it make a backup of translations.cfg and config.cfg like "config.cfg_datetime" in a folder. That way you can go in and see previous setups, like GPIO setup, Template link, Plugin Link - for faster re.setup

anxuae, 2021-05-01:
Creating back files will not solve the main issue which is keeping the old configuration. Obviously, if you reset two times, your backup files will be also erased.

Another way to manage it: add ability to select the config directory at startup.

Something like:

```
$> pibooth ~/Documents/my_config
```

DJ-Dingo, 2021-05-02:
@anxuae  Yes you right. Not that you allways need the same config setup, and sometime you reset if you manual make changes in config that making a crash or something. 

So directory is maybe a good idea, or config backup files with date-time at the end of filename to prevent it getting erased.

anxuae, 2021-05-06:
I add made a first implementation by adding an option to chose the config directory at startup, see PR #236

anxuae, 2021-05-07:
Available in `pibooth` v2.0.2

BR
anxuae

closed 2021-05-07 by anxuae