pibooth/pibooth #233: Issue with taking more that one photo

oldspinner, 2021-05-02:
[I am having an issue with taking more than 1 photo to make a collage. The individual photo works fine but every time I try to use the 4 photo option it takes all the photos then says "oops, something went wrong"
I think it may have something do with the "Unsupported option settings/capturetarget=Memory card ([-2] Bad parameters), configure your DSLR manually" message I am getting when I load the app but I cant seem to find a way to fix that either.
I am using a Sony ZV1

Any ideas?
Thanks

[pibooth.log](https://github.com/pibooth/pibooth/files/6410888/pibooth.log)
](url)

anxuae, 2021-05-02:
Hi @oldspinner 

Do you have a memory card in you DSLR camera?
Generally the case appears because the DSLR same the image in its internal memory instead of the memory card. In this case, only the last photo is kept in the internal memory. Thus if you take a sequence of more than one photo, only the last is accessible and `pibooth` crash.

oldspinner, 2021-05-02:
yep looks like this is what is happening, is there a way of changing this or do I have to try with a different camera?

[diagnostic.log](https://github.com/pibooth/pibooth/files/6411086/diagnostic.log)

anxuae, 2021-05-02:
Looking in the diagnostic file, it seems that your camera is currently saving image in memory SD card and ram. Maybe saving only in SD card will solve the issue? Have a look in the settings of your camera...

```
/settings/capturetarget
  Label       : Capture Target
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Radio button
  Current     : card+sdram
  Choices     : ['sdram', 'card+sdram', 'card']
```

oldspinner, 2021-05-02:
is it possible to change the chose within gphoto maybe? looks like this is an issue with sony cameras

anxuae, 2021-05-02:
try in a shell:

```
gphoto2 --set-config capturetarget=card
```

oldspinner, 2021-05-02:
Thank you so much for your help! I seem to be having an issue running anything through gphoto2 now. I think I may have broken something bigger in the process. Thanks again, I have learnt so much

closed 2021-05-02 by oldspinner

stefaang, 2022-02-14:
[This line forces the capturetarget](https://github.com/pibooth/pibooth/blob/2.0.4/pibooth/camera/gphoto.py#L104) to the External SD card. 
It causes pibooth to freeze for 90 seconds on my Canon DSLR when there's no SD card present on `camera.capture`.
Changed it to `Internal Ram` and it works fine now. (any downsides here?)

Maybe it's possible to detect the presence of an SD card before hardcoding to write to it?