pibooth/pibooth #234: Last picture

arthur5640, 2021-05-03:
When you delete footer text 1 and 2, and you put a new overlay, size of picture change ( size between picture and bordure). How i can doing to have the same result with my overlay?

anxuae, 2021-05-04:
Hi @arthur5640 ,

I'm not sure to understand your need. If you don't put any footer text, by default `pibooth` extends the picture on the whole surface of the background. If you want to keep the margins same than with footer texts, you can set empty footers:

```
# Main text displayed
footer_text1 = "  "

# Secondary text displayed
footer_text2 = "  "
```

anxuae, 2021-05-05:
@arthur5640 do I answered to your question?

arthur5640, 2021-05-06:
hello, yes it's ok. 
but i have a problem when i change the overlay. there is a bug , even if i put the same ratio . (3600*2400)

anxuae, 2021-05-06:
Can you exactly describe the bug?

arthur5640, 2021-05-06:
I have a camera pi hq.
I put a picture resolution 3600x2400
I do an overlay ( not background png) with can with the size 3600x2400
When i choose to take for exemple 1 photo, caméra bug. And i can't escape pibooth

anxuae, 2021-05-07:
I never try the HQ camera, thus difficult to help you.
Your resolution is really higher than I never try, maybe this is too much to be processed by the raspberry pi...
Did you try without overlay and a smaller resolution?

anxuae, 2021-05-22:
Hi @arthur5640 
Did you try other resolutions ?

arthur5640, 2021-05-22:
No. Thanks Envoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : sam. 22 mai 2021 à 12:24À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Hi @arthur5640
Did you try other resolutions ?

anxuae, 2021-06-06:
Hi @arthur5640,

After some tests, I can use the HQ Camera in full resolution (4056 x 3040 pixels), by increasing the GPU memory to 256Mb.

```bash
sudo raspi-config
```

Go to `Performance Options` > `GPU Memory` > `256`
Then reboot the Raspberry-Pi.

Can you try and give me your feedback?

arthur5640, 2021-06-06:
Ok thanks, i will tryEnvoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : dim. 6 juin 2021 à 20:37À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Hi @arthur5640,
After some tests, I can use the HQ Camera in full resolution (4056 x 3040 pixels), by increasing the GPU memory to 256Mb.
sudo raspi-config
Go to Performance Options > GPU Memory > 256
Then reboot the Raspberry-Pi.
Can you try and give me your feedback?

arthur5640, 2021-06-09:
Hello, i have test its ok. I have an other problem :When I run a print, sometimes there are two or three copies that come out. Where can the problem come from?Envoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : dim. 6 juin 2021 à 20:37À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Hi @arthur5640,
After some tests, I can use the HQ Camera in full resolution (4056 x 3040 pixels), by increasing the GPU memory to 256Mb.
sudo raspi-config
Go to Performance Options > GPU Memory > 256
Then reboot the Raspberry-Pi.
Can you try and give me your feedback?

arthur5640, 2021-06-09:
I use touchEnvoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : mer. 9 juin 2021 à 15:15À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Generally is coming from bad contact from the hardware button, you can change the debounce_delay in the config to try to fix: increasing the time means that you need to press longer on the button in order to trigger it.
Other possibility is to change your hardware button for a better quality 😄

arthur5640, 2021-06-09:
Is not possible to put only one copy of print? Envoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : mer. 9 juin 2021 à 15:15À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Generally is coming from bad contact from the hardware button, you can change the debounce_delay in the config to try to fix: increasing the time means that you need to press longer on the button in order to trigger it.
Other possibility is to change your hardware button for a better quality 😄

anxuae, 2021-06-09:
So if you are using touch that means the sensibility of the touch screen is too high, thus several print event are generated.
You can set the max printed duplicate in the config by setting `max_duplicates=1`

arthur5640, 2021-06-09:
Thank you so much. One last question.I chose to take one or two photos. Is it possible by the classic options to define two different wallpapersEnvoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : mer. 9 juin 2021 à 16:32À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
So if you are using touch that means the sensibility of the touch screen is too high, thus several print event are generated.
You can set the max printed duplicate in the config by setting max_duplicates=1

arthur5640, 2021-06-09:
Yes but it is possible to have a different background, if i choose one or four picture for example? When i put an png, it's the same if i take one or many pictures Envoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : mer. 9 juin 2021 à 19:40À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Yes of course, you can set the config key backgrounds with a list of path to different background images.
Have a look to https://github.com/pibooth/pibooth/blob/master/docs/config.rst , you will get all the existing configuration options.

arthur5640, 2021-06-09:
Can you send me an example? Please Envoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : mer. 9 juin 2021 à 20:00À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Put a list of png in the key

anxuae, 2021-06-09:
```
pibooth --config
```

got to the section `[PICTURE]`

```
backgrounds = ("backgrounds/bg1.jpg", "backgrounds/bg2.jpg")
```

of course, change the line to the correct path

arthur5640, 2021-06-09:
Very good thanks Envoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : mer. 9 juin 2021 à 22:02À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
pibooth --config

got to the section [PICTURE]
backgrounds = ("backgrounds/bg1.jpg", "backgrounds/bg2.jpg")

of course, change the line to the correct path

anxuae, 2021-06-09:
Your welcome.
I'm proposing to close this issue because there are lots of topics covered.
Please open new issue per problem.

closed 2021-06-09 by anxuae

arthur5640, 2021-06-09:
Ok i understand. Thanks a lotEnvoyé depuis mon téléphone Orange-------- Message original --------De : Anxuae ***@***.***>Date : mer. 9 juin 2021 à 22:38À : pibooth/pibooth ***@***.***>Cc : arthur5640 ***@***.***>, Mention ***@***.***>Objet : Re: [pibooth/pibooth] Last picture  (#234)
Your welcome.
I'm proposing to close this issue because there are lots of topics covered.
Please open new issue per problem.

anxuae, 2021-06-10:
Do you have put quotes around paths?
Please attach your configuration file