pibooth/pibooth #237: Default to X number of photos so only require one button press required

AndPDale, 2021-05-08:
Hi PiBooth Team

**Describe the solution you'd like**
Is it possible for the number of photographs taken to default to X number defined in config, so that only one button is required to be pressed from the home screen, rather than getting the user to select another button?

**Current steps:**
Home Screen (press button) > Choose 1 photo or x number (press another button to choose) > photos > resulting photos
**Proposed idea:**
Home Screen (press button) > Pose screen > take X number of photos > Resulting photos

werdeil, 2021-05-08:
Hi @AndPDale , this shall already be possible, simply change the config to have only choice for example 
```
captures = 4
```

AndPDale, 2021-05-08:
Thanks @werdeil  for the prompt answer. 
I should have identified that, i just did not scroll through the numbers enough.

closed 2021-05-08 by AndPDale

werdeil, 2021-05-08:
No problem happy to see you are using our pibooth