pibooth/pibooth #238: Introduction of a Pose or Get ready! screen before countdown and preview start

AndPDale, 2021-05-08:
Hi PiBooth team,

Would it be possible to introduce a pose or Get Ready! screen before the countdown to allow users to compose themselves before going straight into the countdown?

Proposed idea:
Home Screen (press button) > Pose screen > take X number of photos > Resulting photos

DJ-Dingo, 2021-05-11:
@AndPDale 

I once changed the "Countdown" to 12 sec. After looking at the pictures, i changed it back to 3 sec. 
Problems was that the kids could not understand that it did not record a video (which would be cool), and they pressing the button just to pose and when the pics was taking they where not ready anyway. (Same thing with drunk people later that night) People where on the way out of the screen in the finish picture. 

Plus, if there are many who have to take pictures, people have to stand in line for a long time because some spend too much time with nothing. I found that it was better that they took 5-10 pictures (and quickly learned to use it) where some pics got bad I could just delete them afterwards. A trick is to make a line, with some tape on the floor, where people have to stand behind, to take the best picture

For some people this becomes a playground, and people quickly learn to Pose in seconds. You can try changing the "Countdown" to 5 Sec. 

I guess what you want is to have the camera on, all the time instead of showing the last picture taking.

werdeil, 2021-05-11:
@AndPDale I share @anxuae point of view, we had it at the beginning but we find it better to have a shorten cycle, if the user didn't understood that the image will be taken directly they will have the chance to restart quickly.

anxuae, 2021-05-12:
Hi @AndPDale 

If really you want to display a text before the countdown, here is a small plugin that doing the job. Juste save it in a file like `pibooth_getready_screen.py` and register it in the config (key `[GENERAL][plugins]=pibooth_getready_screen.py`.

Be aware that it tweak a little bit the `pibooth` code and I can not guarantee correct working with future `pibooth` versions.

```python
"""Plugin to display 'Get Ready' at screen after 'wait' state."""

import time
import pygame
import pibooth
from pibooth import fonts

__version__ = "0.0.1"

@pibooth.hookimpl
def state_wait_exit(win):
    win_rect = win.surface.get_rect()
    text = "Get Reatdy!"

    # Get best font size according to window size
    font = fonts.get_pygame_font(text, fonts.CURRENT,
                                 win_rect.width//1.5, win_rect.height//1.5)

    # Build a surface to display at screen
    text_surface = font.render(text, True, win.text_color)
    print(win_rect)

    # Clear screen
    win.surface.fill(win.bg_color)

    # Draw the surface at screen
    win.surface.blit(text_surface, text_surface.get_rect(center=win_rect.center).topleft)

    # Force screen update and events process
    pygame.display.update()
    pygame.event.pump()

    # Wait 1s
    time.sleep(1)
```

AndPDale, 2021-05-12:
@anxuae 
Thank you very much for your comments and advice regarding your previous findings with using a longer photo process.

I am very grateful to you for providing plug-in code to provide this functionality. I will give it a go.

Much appreciated

anxuae, 2021-05-13:
I let you close the issue if everything works fine

AndPDale, 2021-05-13:
@anxuae 
I have added the plugin you developed and confirmed that it appears in the menu and is enabled
When I run pibooth I am receiving the following error

The error is regarding the pages background. I am using an image for my background in the config file as follows:
"
# Background RGB color or image path
background = /home/pi/Pictures/pibooth_resources/background_gradient.jpg
"
Error message:
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.4
[ INFO    ] pibooth           : Installed plugins: extra-lights-1.0.1, sound-effects-0.0.3, getready_screen-0.0.1
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
**[ ERROR   ] pibooth           : invalid color name**
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1403 SMP Mon Feb 22 11:33:35 GMT 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

I have replaced the image path with #ffffff and I see the following error:

pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.4
[ INFO    ] pibooth           : Installed plugins: extra-lights-1.0.1, sound-effects-0.0.3, getready_screen-0.0.1
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ ERROR   ] pibooth           : Invalid image for window background: '/home/pi/.config/pibooth/#ffffff'
[ ERROR   ] pibooth           : invalid color name
[ ERROR   ] pibooth           : Invalid image for window background: '/home/pi/.config/pibooth/#ffffff'
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/states.py", line 91, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/plugins/view_plugin.py", line 52, in state_wait_enter
    and app.count.remaining_duplicates > 0)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/view/window.py", line 224, in show_intro
    self._update_background(background.IntroBackground(self.arrow_location, self.arrow_offset))
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/view/window.py", line 110, in _update_background
    self._current_background.set_color(self.bg_color)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/view/background.py", line 119, in set_color
    assert osp.isfile(color_or_path), "Invalid image for window background: '{}'".format(color_or_path)
AssertionError: Invalid image for window background: '/home/pi/.config/pibooth/#ffffff'

[... 23 lines cut ...]
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/plugins/view_plugin.py", line 31, in state_failsafe_enter
    win.show_oops()
  File "/home/pi/.local/lib/python3.7/site-packages/pibooth/view/window.py", line 215, in show_oops
    self._update_background(background.OopsBackground())
[... cut ...]
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

I will continue to look into this issue, but any advice that you have will be great

Thanks

anxuae, 2021-05-13:
I see, I made the things simple considering you are using a color for your background 😭 

Below is an update of the plugin:

```python
"""Plugin to display 'Get Ready' at screen after 'wait' state."""

import time
import pygame
import pibooth
from pibooth import pictures, fonts

__version__ = "0.0.2"

@pibooth.hookimpl
def state_wait_exit(win):
    win_rect = win.surface.get_rect()
    text = "Get Reatdy!"

    # Get best font size according to window size
    font = fonts.get_pygame_font(text, fonts.CURRENT,
                                 win_rect.width//1.5, win_rect.height//1.5)

    # Build a surface to display at screen
    text_surface = font.render(text, True, win.text_color)

    # Clear screen
    if isinstance(win.bg_color, (tuple, list)):
        win.surface.fill(win.bg_color)
    else:
        bg_surface = pictures.get_pygame_image(win.bg_color, win_rect.size, crop=True, color=None)
        win.surface.blit(bg_surface, (0, 0))

    # Draw the surface at screen
    win.surface.blit(text_surface, text_surface.get_rect(center=win_rect.center).topleft)

    # Force screen update and events process
    pygame.display.update()
    pygame.event.pump()

    # Wait 1s
    time.sleep(1)
```

AndPDale, 2021-05-13:
@anxuae 
Thank you for the prompt reply.
This has now functions correctly and has allowed me to use a background image

closed 2021-05-14 by anxuae