pibooth/pibooth #243: Menu open when button capture is pressed... again....?=?!

Aurel-Charles, 2021-05-13:
Issue already resolved in other thread

_Originally posted by @RAW4RMCS in https://github.com/pibooth/pibooth/issues/175#issuecomment-686803228_

but for some reason it doesn't work for me......

BUG:
once pibooth is at the "start", I press the capture button and then it open the menu insted of going to the "chose" page
also noticed that the print is not doing anything, even if I go to the "chose" page by pressing the key "p" and then pressing the print button.

If in the config file I erase the capture button, then after relaunch (of course that button is dead) but the print button works.
I've also tried changing the debounce button delay on all sort of way (even with the multi-press)
but still no luck..! 

I some of you have any idae, I'll be happy to try!

anxuae, 2021-05-13:
Hello @Aurel-Charles,

I never experienced this behavior. But with some kind of HW buttons I have noticed that if there is too slack between the mobile part and fixed part, unwanted contacts are generated by the button.

Maybe you can try with other model of HW buttons?

DJ-Dingo, 2021-05-13:
I had this problem a year ago, can not remember what exactly i did to solve it
But here are some tricks you can try.

Use some GPIO for the buttons that are not too close together or other used GPIO. This may be Interference.
(Some times change of cables or power supply may help) 

2. Turn off bluetooth / Wifi if you are not using it. (And don´t have your router or mobile phone close to the Raspberry Pi)

3. Use some Ferrite around the cables. This I have done alot to my projects to prevent Interference.

https://www.google.com/search?q=ferrite&rlz=1C1CHBD_daNL872NL872&sxsrf=ALeKk02JlQAGh40mz8hbo8sDYA2P5K2GmA:1620934001525&source=lnms&tbm=isch&sa=X&ved=2ahUKEwiYyfXascfwAhUNshQKHccuDPgQ_AUoAXoECAEQAw&biw=1280&bih=638#imgrc=iQKSDZoY3XVdYM

Aurel-Charles, 2021-05-14:
ok guys!! I've change my HW button and now it works! 
thanks for your help!
great jobs on the V.2.0.2 !!

closed 2021-05-14 by Aurel-Charles

Schillers, 2022-04-26:
> ok guys!! I've change my HW button and now it works! thanks for your help! great jobs on the V.2.0.2 !!

@Aurel-Charles , would you mind if I asked which buttons you were using and which buttons you moved to? I have the same issue!