pibooth/pibooth #244: Canon EOS60D not working with last version

thibault-78, 2021-05-14:
Hello,
With last version 2.0.2 
My DSLR cam not working anymore alone
Problem is linked to change: #Add hybrid camera OpenCV+gPhoto2 #158
At init when I have only DSLR cam connected the init go through Hybrid openCV / Photo2
-> USB locked
-> I think because my EOS60D can be detected by the 2 software
I bypassed the Hybrid directly in the init and working

anxuae, 2021-05-14:
Yes you have probably right, can you post me the trackback of the crash?
I will try to fix it in the master branch.

anxuae, 2021-05-15:
Hi @thibault-78 

Can you test the PR #245 with your setup?

```bash
$> git clone https://github.com/pibooth/pibooth.git
$> cd pibooth
$> git checkout camera
$> git pull
$> pip install -e .[dslr,printer]
$> pibooth
```

anxuae, 2021-05-17:
After some tests, it should be fixed by #245

thibault-78, 2021-05-18:
Hello @anxuae 
I've just try the new code
Init is OK in DSLR mode
-> pibooth working :)
Init is now not OK in hybrid mode OpenCV+GP2
Here is the trace
```
pibooth
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.4
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring hybrid camera (OpenCV + gPhoto2) ...
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    load_entry_point('pibooth', 'console_scripts', 'pibooth')()
  File "/home/pi/pibooth/pibooth/booth.py", line 471, in main
    app = PiApplication(config, plugin_manager)
  File "/home/pi/pibooth/pibooth/booth.py", line 105, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 203, in _multicall
    gen.send(outcome)
  File "/home/pi/pibooth/pibooth/plugins/camera_plugin.py", line 26, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/home/pi/pibooth/pibooth/camera/__init__.py", line 38, in find_camera
    return HybridCvCamera(cv_cam, gp_cam)
  File "/home/pi/pibooth/pibooth/camera/hybrid.py", line 60, in __init__
    super(HybridCvCamera, self).__init__(cv_camera_proxy)
  File "/home/pi/pibooth/pibooth/camera/opencv.py", line 59, in __init__
    if not isinstance(camera_proxy, cv2.VideoCapture):
TypeError: isinstance() arg 2 must be a type or tuple of types
```

thibault-78, 2021-05-18:
Other test:
init crash also in OPenCV mode only

```
aspberrypi:~/pibooth $ pibooth
pygame 2.0.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.4
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring OpenCV camera ...
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    load_entry_point('pibooth', 'console_scripts', 'pibooth')()
  File "/home/pi/pibooth/pibooth/booth.py", line 471, in main
    app = PiApplication(config, plugin_manager)
  File "/home/pi/pibooth/pibooth/booth.py", line 105, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 203, in _multicall
    gen.send(outcome)
  File "/home/pi/pibooth/pibooth/plugins/camera_plugin.py", line 26, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/home/pi/pibooth/pibooth/camera/__init__.py", line 50, in find_camera
    return CvCamera(cv_cam)
  File "/home/pi/pibooth/pibooth/camera/opencv.py", line 59, in __init__
    if not isinstance(camera_proxy, cv2.VideoCapture):
TypeError: isinstance() arg 2 must be a type or tuple of types
```

My cam is A logitech

anxuae, 2021-05-18:
Hi @thibault-78 ,

Thanks for testing, I just push a fix on the `camera` branch, can you do a pull and recheck?

thibault-78, 2021-05-19:
Hello
Just made some tests:
In GP2 only = OK
In Hybrid OpenCV + GP2 = OK
In OpenCV = OK
Many thanks for your work

anxuae, 2021-05-22:
Fixed in v2.0.3

closed 2021-05-22 by anxuae