pibooth/pibooth #246: Canon EOS 60D Autofocus not focus => Oops

thibault-78, 2021-05-18:
with version 2.0.2
in Gphoto2 mode with canon EOS 60D
if autofocus of the camera can not make the focus
it result a "Oops something.""

```
[ INFO    ] pibooth           : +< Take a capture
[ DEBUG   ] pibooth           : |  Setting option actions/viewfinder=0
[ DEBUG   ] pibooth.gphoto2   : |  ptp_usb_getresp [usb.c:514]: PTP_OC 0x9128 receiving resp failed: PTP Device Busy (0x2019)
[ DEBUG   ] pibooth.gphoto2   : |  camera_trigger_canon_eos_capture [library.c:5015]: 'ptp_canon_eos_remotereleaseon (params, 2, 0)' failed: Canon EOS Full-Press failed (0x2019: PTP Device Busy)
[ DEBUG   ] pibooth.gphoto2   : |  gp_context_error: Canon EOS Full-Press failed (0x2019: PTP Device Busy)
[ DEBUG   ] pibooth.gphoto2   : |  camera_canon_eos_capture [library.c:3675]: 'camera_trigger_canon_eos_capture (camera, context)' failed: 'Unknown error' (-110)
[ DEBUG   ] pibooth.gphoto2   : |  gp_camera_capture [gphoto2-camera.c:1340]: 'camera->functions->capture (camera, type, path, context)' failed: -110
[ DEBUG   ] pibooth           : +> took 1.256 seconds
[ ERROR   ] pibooth           : [-110] I/O in progress
Traceback (most recent call last):
  File "/home/pi/pibooth/pibooth/booth.py", line 377, in main_loop
    self._machine.process(events)
  File "/home/pi/pibooth/pibooth/states.py", line 50, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/pibooth/pibooth/plugins/camera_plugin.py", line 100, in state_capture_do
    app.camera.capture(effect)
  File "/home/pi/pibooth/pibooth/camera/gphoto.py", line 298, in capture
    self._captures.append((self._cam.capture(gp.GP_CAPTURE_IMAGE), effect))
gphoto2.GPhoto2Error: [-110] I/O in progress
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1414 SMP Fri Apr 30 13:20:47 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
```

Not a big deal as we can start again to make another picture
-> in case of multiple Picture:  if it fails on the last we loose all the photos taken before

If not possible to handle that issue (probably more linked to Gphoto2), would it be possible to
-  tell to the "user" to not be too close of the camera. 
- maybee just restart the last picture and not the all previous one of the series

werdeil, 2021-05-18:
Hi @thibault-78, this was in fact the first reason we implemented the Oops page so I don't think we can consider this as a  bug.

About telling the user to not be too close: it is tricky as the I/O error from Gphoto can come from multiple reasons
About restarting the last picture: we can enter an inifinite loop so I don't think we should implement it either.

anxuae, 2021-05-19:
Moreover, you can customize the "Oops" message if you want a different text (`pibooth --translate`), but be aware that your message could be in discordance with the real error.

anxuae, 2021-05-22:
Hi @thibault-78 ,

Using:
- `--translate` to customize the message
- new hook `pibooth_setup_camera` to customize the camera (see https://github.com/pibooth/pibooth/blob/master/docs/plugin.rst#example-5--setup-a-custom-camera)

You should obtain the features that you want, thus I close this issue

closed 2021-05-22 by anxuae