pibooth/pibooth #247: Sony Alpha SLT-58 did not work

TeyBay, 2021-05-20:
**Describe the bug**
Pibooth crashes at taking a photo, gphoto alone works well with the camera.

**Steps to reproduce the bug**

**Logs**
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring hybrid camera (OpenCV + gPhoto2) ...
[ WARNING ] pibooth           : The connected DSLR camera is not compatible with preview
[ DEBUG   ] pibooth           : Setting option imgsettings/iso=100
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Bulb Mode' expected: 0x0000 got: 0x0004
[ DEBUG   ] pibooth           : Setting option settings/capturetarget=Memory card
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Bulb Mode' expected: 0x0000 got: 0x0004
[ ERROR   ] pibooth           : Unsupported option settings/capturetarget=Memory card ([-2] Bad parameters), configure your DSLR manually
[ DEBUG   ] pibooth.gphoto2   : gp_libusb1_open [libusb1.c:431]: 'libusb_claim_interface (port->pl->dh, port->settings.usb.interface)' failed: Resource busy (-6)
[ DEBUG   ] pibooth.gphoto2   : gp_port_set_error [gphoto2-port.c:1190]: Could not claim interface 0 (Success). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
[ DEBUG   ] pibooth.gphoto2   : gp_port_set_error [gphoto2-port.c:1190]: Could not release interface 0 (Success).
[ DEBUG   ] pibooth.gphoto2   : gp_port_set_error [gphoto2-port.c:1190]: Could not release interface 0 (Success).
[ DEBUG   ] pibooth           : Preview resolution is (0.0, 0.0)
[ INFO    ] pibooth           : Connected to printer 'Epson_Stylus_Office_BX535WD'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 0.345 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : took 0.181 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 382, in main_loop
    self._machine.process(events)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 64, in process
    self.set_state(new_state_name)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 91, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 337, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 52, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 335, in <lambda>
    outcome = _Result.from_call(lambda: oldcall(hook, hook_impls, kwargs))
  File "/usr/local/lib/python3.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/camera_plugin.py", line 53, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/opencv.py", line 150, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/opencv.py", line 96, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1414 SMP Fri Apr 30 13:20:47 BST 2021', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
**Environment**
 - OS Raspbian
 - Camera Sony SLT-A58
 - Pibooth version 2.0.1

anxuae, 2021-05-21:
Hi @TeyBay, I think you are on the `pibooth` 2.0.2 version, can you confirm?
This issue will be closed in `pibooth` 2.0.3.

TeyBay, 2021-05-21:
Yes. When will be 2.0.3 available?

anxuae, 2021-05-22:
Please retry with v2.0.3.
Reopen a new issue if there are other problems

closed 2021-05-22 by anxuae