pibooth/pibooth #248: Preview don't work with HQ camera

lolotan, 2021-05-21:
**Bug description**
I'm unable to have the preview before taking a picture.
All other functionalities works perfectly (the picture is taken at default config resolution).
I tested the camera preview with raspistill (`raspistill -t 0`) and picamera library (`picamera.start_preview()`) and the preview works perfectly in both case, so it doesn't seems to be a picamera compatibility issue.

**Environment**
 - OS: Raspberry Pi OS (Buster)
 - Camera: Raspberry Pi HQ camera (Sony IMX477 / CSI)
 - Pibooth version: 2.02 and master (774c059d40322f6661799770aeb05ab34ca58789)
 - Tested on Raspberry Pi 3A+ and 3B+ with gpu_mem=128, gpu_mem=256 and even gpu_mem=512 (on 3B+)

anxuae, 2021-05-21:
Hi @lolotan ,

Can you confirm me tat you are not running `pibooth` over SSH?
Please read carefully the https://github.com/pibooth/pibooth#requirements warning.

lolotan, 2021-05-21:
Hi @anxuae,
I confirm pibooth is not running over SSH.
I execute pibooth with ssh for testing purposes, and I add a line on rc.local for running it at startup (the autostart option don't work also).

Here is my configuration:
- Raspberry Pi 3B+
- HQ camera
- Old 17" 4/3 screen (1280x1024) with DVI to HDMI adapter
- CP1300 photo printer
- Two push buttons with LEDs

As I said: running preview from raspistill or picamera/python works perfectly with this setup (I set the focus thanks to the preview).

Thanks a lot.

anxuae, 2021-05-21:
Have you tried running `pibooth` after raspberry pi startup, from a shell opened in the desktop (i.e. calling it directly than via rc.local)?

lolotan, 2021-05-21:
Thanks for your answers,

> Have you tried running `pibooth` after raspberry pi startup, from a shell opened in the desktop (i.e. calling it directly than via rc.local)?
Yes, I've just tried that, without success unfortunately :/

> Note: I'm running `pibooth` on the latest `Pi OS with desktop` and `pibooth` starts correctly at Raspberry Pi startup without add it to `rc.local`. I've tried on Buster and works also.
I also tried with the latest Pi OS, but the lite version. It may worth the try with the complete version including desktop (if I have the time, because it must be done for this week end).

> Try starting the preview in a python shell, maybe the parameters are not compatible?
> (Sorry but I have no HQ camera to test by myself)
> 
> ```python
> # Take (width, height) from [WINDOW][size] in config
> 
> picamera.start_preview(resolution=(width, height), hflip=False, fullscreen=False)
> ```
I'll try that as soon as I can, an I keep you in touch.
Thanks again.

anxuae, 2021-06-06:
Hi @lolotan,

After some tests, I can use the HQ Camera in full resolution (4056 x 3040 pixels), by increasing the GPU memory to 256Mb.

```bash
sudo raspi-config
```

Go to `Performance Options` > `GPU Memory` > `256`
Then reboot the Raspberry-Pi.

Can you try and give me your feedback?

lolotan, 2021-06-07:
Ho @anxuae,
Sorry for the lack of answer, I had to dismantle my photobooth after the event, and I can't manage to make it work again unfortunately.
I had already increased the GPU memory (I switched from a 3A+ to a 3B+ for that), and as far as I know, I didn't have any problem capturing pictures. I was just unable to have a preview, whatever configuration or options I could set.

Feel free to close the ticket, if no one else has this issue.

anxuae, 2021-06-09:
The problem seems to be solved by increasing GPU memory allocation (see #234)

closed 2021-06-09 by anxuae