pibooth/pibooth #249: VERY slow keyboard reaction in, v2.0.2 and v2.0.3

DJ-Dingo, 2021-05-22:
Slow keyboard reaction.

If you need to type in some text, for example "Title / Sub-Title, or in the LCD pluging", the keyboard response is VERY slow. You must press a key and wait 1-2 sec. before pressing the next letter.

I just updatet to V2.0.3, and it´s the same.
(It was by the way standing still for 1 minute, before it continued and starting updating) ?

anxuae, 2021-05-24:
Hi @DJ-Dingo,

I'm never experienced this behavior but I've noticed that the touch area a little bite slow with 4.x versions of https://github.com/ppizarror/pygame-menu .

Could you tell me if with version 2.0.1 or 2.0.0 was better?
Slow reaction is only for input text of for the whole menu actions?

werdeil, 2021-05-24:
And to add to @anxuae questions : do you have issues with a physical keyboard or a virtual one on the touchscreen?

DJ-Dingo, 2021-05-24:
@werdeil - physical keyboard

@anxuae - 
> Could you tell me if with version 2.0.1 or 2.0.0 was better?

Yes both version 2.0.1 and 2.0.0 was better. 
It is okay in the menu to start with, but as soon i write something it is slow, and after that then the menu is getting slow too.

DJ-Dingo, 2021-05-28:
@anxuae 
Just found out, that when i press "Esc" on the keyboard, it takes 4-10 Sec. before the menu pops up. (Keyboard works normal in etc. a text-editor or browser). 

Since it is only in Pybooth, when i writes something i need to press the keys for longer time 1-2 sec, (i cant just release it fast). Can this be somekind of update to Pygame-menu 4.0.6 that i miss, or something else i should look at.

ppizarror, 2021-05-28:
Hi @DJ-Dingo, I'm aware that my library is being used in this project. Many people have used pygame-menu with rapsberry's and no issues have been raised. 

Can you test the menu directly? There's a couple of examples you could try.
https://pygame-menu.readthedocs.io/en/4.0.6/_source/gallery.html

Therefore we can check if there's an issue regarding pibooth implementation, or is because pygame-menu. Maybe there's an issue about the menu updating .. but I did not modify any regarding the update or drawing mechanism (from v4.0.4 to v4.0.6).

DJ-Dingo, 2021-05-29:
Hi @ppizarror, Cool thx for the info, i will do some testing and let you know.

DJ-Dingo, 2021-05-29:
Hi @ppizarror, 

I have now tested and it works fine, there is a little bit of a delay when I type fast or use the arrow keys fast, but it is very very little and it only forgot one letter sometimes when i type very fast, and not at all as much as Pibooth does.

Had to use "python3", or it would not work at all
python3 -m pygame_menu.examples.simple

Since no one else has experienced it, maybe I have some different update conflicts and the way something is installed in the past.

ppizarror, 2021-05-29:
Nice, regarding keyboard <data race> I know that is a problem, and in the future, the key conflict (pressing many times an event) must be solved. Also, let me know if there's any issue regarding the menu on my Github repo :)

anxuae, 2021-06-09:
Hi @DJ-Dingo ,

Did you try to re-install the OS? If yes do you still have the issue?

DJ-Dingo, 2021-06-09:
@anxuae 
Not yet. I will try that on a new sd card soon, i let you know if that helps. 
But still like to figure out why it is doing it.

Do you want me to close this ?

anxuae, 2021-06-09:
I'm agree with you. We can keep this issue opened until we found the reason.

anxuae, 2023-05-06:
Hi @DJ-Dingo,

I'm closing this old issue because no more occurence. Please re-open if needed.

closed 2023-05-06 by anxuae