pibooth/pibooth #250: Touchscreen question.

DJ-Dingo, 2021-05-22:
I have a Dell e2014T touch screen, mainly used it because it has a protective glass, and children like to point and touch the screen.

But how does the touchscreen work. I have not been able to get it to respond at all. I know it worked in the old days on some "Raspberry pi 2" projects but have not used it since. 

The question is whether this monitor is supported at all in the newer versions of Raspberry Pi OS., Or do I need to do something specific to make it work with Pibooth ?

anxuae, 2021-05-24:
Hi @DJ-Dingo 

`pibooth` is built on to of `pygame` library, thus if your screen work with pygame, it will work with pibooth.
However I never try to do that. On this website, they use the same screen than you on a Raspberry Pi: https://projects-raspberry.com/digital-music-stand-raspberry-pi-touchscreen : have a look at `step 2`.

If you connect your screen, does the touches work (for instance clicking on the menu bar of the pi desktop)?

werdeil, 2021-05-24:
And to add to @anxuae questions, do you have issue with a physical keyboard or with a touchscreen and the virtual keyboard?

DJ-Dingo, 2021-05-24:
> If you connect your screen, does the touches work (for instance clicking on the menu bar of the pi desktop)?

No, it dont. But i give step 2 a try again. :-)

DJ-Dingo, 2021-05-25:
Gonna close this for now, since it has nothing to do with Pibooth :-)

I am still trying to get it to wotk, but without any success yet.

closed 2021-05-25 by DJ-Dingo